ARMmbed/mbed-os #7949: Can't compile in MCUXpresso IDE 10.2 (Make Error 126, Bad file number)

maclobdell, 2018-08-31:
### Description

After exporting an Mbed OS project to MCUXpresso IDE, I open it, click build, and consistently get Make Error 126.  See below.  This happens on different files in different projects.  

I've tried opening MCUXpresso as an administrator.  Tried re-importing projects, moving their location up closer to C:/ , building from within the workspace or outside of it, etc.  

IDE tested : MCUXpresso IDE 10.2.1_795
Targets tested: K64F, LPC546XX
Mbed OS Versions tested: 5.9.0 & 5.9.6
Mbed OS applications tested: mbed-os-example-blinky, simple-cloud-client-example
Host OS: Windows 10

```
16:05:26 **** Build of configuration Debug for project mbed-os-example-blinky ****
make -r -j4 linker-script-debug.ld all 
arm-none-eabi-cpp -E -P -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -Wl,-n -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys  -Wl,--end-group ../mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld -o linker-script-debug.ld
Building file: ../mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/analogout_api.c
Building file: ../mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/dma_api.c
Building file: ../mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/analogin_api.c
Invoking: MCU C Compiler
Invoking: MCU C Compiler
arm-none-eabi-gcc -std=gnu99 -DTARGET_KSDK2_MCUS -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DMBED_BUILD_TIMESTAMP=1535747969.36 -DDEVICE_CRC=1 -DARM_MATH_CM4 -DTARGET_DEBUG -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SPI_ASYNCH=1 -DDEVICE_LPTICKER=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_Freescale_EMAC -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -DDEVICE_USTICKER=1 -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_FRDM -DTARGET_MCUXpresso_MCUS -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DTARGET_CORTEX_M -D__NEWLIB__ -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/." -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/cmsis" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/cmsis/TARGET_CORTEX_M" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/drivers" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/events" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/events/equeue" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/easy_cellular" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/API" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/AT" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/common" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/QUECTEL" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/QUECTEL/BC95" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/QUECTEL/BG96" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/QUECTEL/UG96" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/TELIT" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/TELIT/HE910" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/UBLOX" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cellular/framework/targets/UBLOX/PPP" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/cryptocell" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/device_key" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/device_key/source" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/filesystem" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/filesystem/bd" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/filesystem/fat" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/filesystem/fat/ChaN" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/filesystem/littlefs" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/filesystem/littlefs/littlefs" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/greentea-client" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/greentea-client/greentea-client" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/greentea-client/source" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-client-randlib" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-client-randlib/mbed-client-randlib" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-client-randlib/mbed-client-randlib/platform" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-client-randlib/source" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-coap" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-coap/doxygen" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-coap/mbed-coap" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-coap/source" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-coap/source/include" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/features/frameworks/mbed-trace" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example- [... cut ...]
make: *** [mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/analogout_api.o] Error 126
/bin/sh: /c/NXP/MCUXpressoIDE_10.2.1_795/ide/tools/bin/arm-none-eabi-gcc: Bad file number
make: *** [mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/dma_api.o] Error 126

16:05:29 Build Finished (took 2s.991ms)
```

I plan to revert to an old version of MCUXpresso IDE and try that out. 

Does anyone know what this problem is caused by??

### Issue request type

    [] Question
    [ ] Enhancement 
    [X] Bug

JojoS62, 2018-09-01:
it looks like mbed became too fat, too many source files for the command line.
Try to exclude all features from the build: right click on mbd-os/features, select 'Resource configurations / Exclude from build...' and select all.
A better solution will be to remove freatures by the mbed configuration option before exporting, but I have not tried this yet.

maclobdell, 2018-09-04:
@JojoS62 - thanks for the great tip.  unfortunately, excluding resources from within MCUXpresso doesn't seem to affect the build.  But if I delete some files manually, it is able to complete the build.  I'll investigate more.

JojoS62, 2018-09-04:
Excluding should work also, but you may have to delete the Debug/Release directories before building because the linker will still find the compiled objectfiles.
Another way is also useful for me:
- add the features that you want to remove to the .mbedignore file in the project root
- export with the -z option to get a zipped projectfile. This will contain only the required files.
Unfortunately, this zip option (don't know it it already official published) has also some problem, but the workaround that I reported in the issue works #7954  for me.

dhearn, 2018-09-11:
In the example by the original poster, the full path to every sub-directory in the source is being passed in multiple times into the command - I think when defining the INCLUDE paths:

`-I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/." -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/cmsis" -I"C:/Users/maclob01/Documents/MCUXpressoIDE_10.2.1_795/workspace/mbed-os-example-blinky/mbed-os/cmsis/TARGET_CORTEX_M" `

As a result, I think it's not (just?) that mbed-os has grown - but it's also affected by the base path to the sources being built, and that's why it's exceeding the command line length.  I had exactly the same problem on my machine.  I found that excluding bits of mbed-os didn't seem to have any effect, however, shortening the path where the code was stored allowed it to build progress much further through the build, before it failed for a legitimate and unrelated compilation error.

So, core problem seems to be that the maximum command line length is being exceeded - this will be influenced by the number of subfolders in the project, as this gets added to the INCLUDE lists, and the path to where the sources are located.

JojoS62, 2018-09-11:
yes, the call to gcc produces a cmd length >36k characters in this example. There are about 240 (!) include path for every feature added.  There should be a confiuration system where I can select the wanted features. If every new feature is automatically added, then a project may fail in the future when the features increase. I cannot explicitly ignore unwanted features if I don't know how it will be named in the future.

dhearn, 2018-09-11:
Now, having got past this earlier issue (Bad file number/error 126 on compiling), I'm also having a similar problem when linking, due to the 32k limit - the link command line is 32144 bytes long:

```
Invoking: MCU C++ Linker
arm-none-eabi-c++ -o "mbed-cli-tech-discovery.axf"  ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/analogin_api.o ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/analogout_api.o ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/dma_api.o ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/flash_api.o ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/gpio_api.o ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/gpio_irq_api.o (...for every .o file in mbed-os and my project...)  ./main.o   -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -Wl,-n -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys  -Wl,--end-group -Xlinker -Map="mbed-cli-tech-discovery.map" -Xlinker -print-memory-usage -Xlinker --gc-sections -Xlinker --wrap=main -Xlinker --wrap=_malloc_r -Xlinker --wrap=_free_r -Xlinker --wrap=_realloc_r -Xlinker --wrap=_memalign_r -Xlinker --wrap=_calloc_r -Xlinker --wrap=exit -Xlinker --wrap=atexit -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb -T C:/Temp/proj/mbed-cli-tech-discovery/Debug/linker-script-debug.ld
arm-none-eabi-c++.exe: error: CreateProcess: No such file or directory
```

This one cannot be easily fixed by changing the overall folder path - as most of the repeated paths are mbed-os ones.

JojoS62, 2018-09-11:
for the linker, it should help to reduce the muber of features by exluding in the .mbedignore file (before exporting) or by exluding the source folders from build (right click on folder in project tree, source / exclude from build).

dhearn, 2018-09-11:
I have already excluded as many features as possible from within mbed-os via the .mbedignore file before exporting.  The features which are left appear to be the minimum are required to use mbed-tls with Ethernet interface, i.e. if I remove any more, it fails to build via `mbed compile`.  I think I based my test code upon https://github.com/ARMmbed/mbed-os-example-tls

JojoS62, 2018-09-24:
I think shortening the Include pathes is easy. In the exporter templates, the include path is prefixed with "${ProjDirPath}/" which expands to the full include path. Changing this to "../" will shorten the path a lot, depending on the workspace location. The working directory for the build is usually always project/Build, so the relative path will work for that.
This applies to many eclipse based exporters, I can it for MCUXpresso and gnuarmeclipse later.
The problem with many linker files will remain. The linker command line can be modified to use a response file instead of adding all object files, but this response file has to be created during compile or before linking.

maclobdell, 2018-09-26:
thanks for the great suggestions.  these are helpful workarounds.  I'd like to see if it is possible to have an architectural solution that makes exporting to and IDE seamless.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-405*