ARMmbed/mbed-os #7954: Exporter: using -z (zip) options fails

JojoS62, 2018-09-02:
### Description
there is a new (undocumented?) function for the exporters to create the export in a zipfile. Great!
but using this option throws an error with traceback:
```
D:\Projects\Sn\LocalGit\mbed-os-example-blinky>mbed export -i make_gcc_arm -z
[mbed] Auto-installing missing Python modules...
Scan: mbed-os-example-blinky
Scan: FEATURE_BLE
Scan: FEATURE_BOOTLOADER
Traceback (most recent call last):
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\project.py", line 379, in <module>
    main()
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\project.py", line 371, in main
    ignore=options.ignore
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\project.py", line 152, in export
    ignore=ignore
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\export\__init__.py", line 287, in export_project
    resources.add_features(ALLOWED_FEATURES)
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\resources\__init__.py", line 236, in add_features
    self._add_labels("FEATURE", features)
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\resources\__init__.py", line 226, in _add_labels
    self.add_directory(path, base_path, into_path)
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\resources\__init__.py", line 423, in add_directory
    self._add_file(file_path, base_path, into_path)
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\resources\__init__.py", line 465, in _add_file
    self.add_file_ref(file_type, fake_path, file_path)
  File "D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\resources\__init__.py", line 258, in add_file_ref
    self._file_refs[file_type].add(ref)
AttributeError: 'list' object has no attribute 'add'
[mbed] ERROR: "C:\Python27\python.exe" returned error code 1.
[mbed] ERROR: Command "C:\Python27\python.exe -u D:\Projects\Sn\LocalGit\mbed-os-example-blinky\mbed-os\tools\project.py -i make_gcc_arm -m STM32F407VE_BLACK --source . -z" in "D:\Projects\Sn\LocalGit\mbed-os-example-blinky"
```

This happens for all exporters and targets that I have tried (LPC1768, BLUEPILL_F103C8, make_gcc_arm, mcuxpresso, gnuarmeclipse).

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

JojoS62, 2018-09-02:
I have added a try/except to skip over the error and to print the ref that is causing the problem:

```
    def add_file_ref(self, file_type, file_name, file_path):
        if sep != self._sep:
            ref = FileRef(file_name.replace(sep, self._sep), file_path)
        else:
            ref = FileRef(file_name, file_path)
        try:
            self._file_refs[file_type].add(ref)
        except:
            print(ref)
            pass
```

This workaround skips the errors and creates the zip files.

```
D:\Projects\Sn\LocalGit\mbed-os-example-blinky>mbed export -i make_gcc_arm -z
[mbed] Auto-installing missing Python modules...
Scan: mbed-os-example-blinky
Scan: FEATURE_BLE
Scan: FEATURE_BOOTLOADER
FileRef(name='mbed-os/features/FEATURE_BOOTLOADER/mbed_lib.json', path='.\\mbed-os\\features\\FEATURE_BOOTLOADER\\mbed_lib.json')
Scan: FEATURE_LWIP
FileRef(name='mbed-os/features/deprecated_warnings/FEATURE_LWIP/feature_lwip_is_deprecated.c', path='.\\mbed-os\\features\\deprecated_warnings\\FEATURE_LWIP\\feature_lwip_is_deprecated.c')
Scan: FEATURE_NANOSTACK
FileRef(name='mbed-os/features/deprecated_warnings/FEATURE_NANOSTACK/feature_nanostack_is_deprecated.c', path='.\\mbed-os\\features\\deprecated_warnings\\FEATURE_NANOSTACK\\feature_nanostack_is_deprecated.c')
Zipping [  0.3%]: .mbed
Zipping [  0.5%]: GettingStarted.html
Zipping [  0.6%]: Makefile
...
```

JojoS62, 2018-09-04:
I‘m using the current master, the zip option is not very old.
My workaround code refers to the error position in the traceback:
mbed-os\tools\resources\__init__.py", line 258, in add_file_ref

theotherjimmy, 2018-09-24:
@JojoS62 That's all correct info. It's really, really odd that it only affect 3 files.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-406*

JojoS62, 2018-10-08:
@theotherjimmy I have seen the zip option is also available in the mbed2 tools. There I got also an error because a notify object is empty, maybe this is only set for the web export? Uncommenting that line 145 in project.py helped:

```
    if targets_json:
        # notify.info("Using targets from %s" % targets_json)
        set_targets_json_location(targets_json)
```

PR #8248 (2018-10-11): Tools, Resources: Convert to a set during win2unix
### Description

The exporters were failing as some things would end up as a list after 
win2unix. Resolves #7954

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-11 by cmonr