ARMmbed/mbed-os #7956: Netsocket API requires RTOS

janjongboom, 2018-09-03:
### Description

features/netsocket uses `Mutex` instead of `PlatformMutex` and thus cannot be used when RTOS is not loaded. In addition it uses `EventFlags` but that can be `ifdef`'ed out.

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

kjbracey-arm, 2018-09-04:
I was under the impression that the RTOS dependency was deeper than that, and hence didn't think Mutex versus PlatformMutex was an issue. Are you suggesting you can get some useful functionality with those changes made?

cmonr, 2018-09-05:
@janjongboom Gonna need some clarification on that description. Those sentences sound like statements, not questions or requests.

janjongboom, 2018-09-06:
@cmonr The request is to make netsocket not dependent on RTOS (at least UDPSocket and TCPSocket). I think it's pretty straightforward, just by swapping out `Mutex` for `PlatformMutex`. The only issue I see is around `EventFlags`. Not sure how they're used but if I comment them out everything still seems to work.

SeppoTakalo, 2018-09-19:
Mbed OS is RTOS.

We have no requirements to support non-RTOS in networking side.

Non RTOS builds are provided for legacy Mbed 2, but Socket API is not supported there.

janjongboom, 2018-09-19:
@SeppoTakalo So why are we even using PlatformMutex then in various places? I think this is much too simplistically thought. 

There is a real use case of using the non-RTOS Mbed OS, whether it's Mbed 2, a constrained target, or a simulated environment. Using the socket APIs is very useful there regardless, and patching the RTOS bits out is in my opinion doable, and thus we should allow that.

@bulislaw

SeppoTakalo, 2018-09-20:
Non-RTOS is completely different operating system from our point of view.

And we have not allocated any resources to support its networking APIs. Maybe the patching seems easy, but I need to take the burden of actually maintaining and testing it.

If such business case appears that requires this, it has to be planned into Mbed OS project and resourced properly.
Until then, I decline this request.

closed 2018-10-02 by adbridge