ARMmbed/mbed-os #7961: `mbed compile --stats-depth=X` is undocumented

SeppoTakalo, 2018-09-03:
### Description

`mbed compile --stats-depth=X` is undocumented

```
$ mbed compile --help
usage: mbed compile [-h] [-t TOOLCHAIN] [-m TARGET] [--profile PROFILE]
                    [--library] [--config] [--prefix CONFIG_PREFIX]
                    [--source SOURCE] [--build BUILD] [-c] [-f] [--sterm]
                    [-N ARTIFACT_NAME] [-S [{matrix,toolchains,targets}]]
                    [--app-config APP_CONFIG] [-v] [-vv]

Compile this program using the mbed build tools.

optional arguments:
  -h, --help            show this help message and exit
  -t TOOLCHAIN, --toolchain TOOLCHAIN
                        Compile toolchain. Example: ARM, GCC_ARM, IAR
  -m TARGET, --target TARGET
                        Compile target MCU. Example: K64F, NUCLEO_F401RE,
                        NRF51822...
  --profile PROFILE     Path of a build profile configuration file. Example:
                        mbed-os/tools/profiles/debug.json
  --library             Compile the current program or library as a static
                        library.
  --config              Show run-time compile configuration
  --prefix CONFIG_PREFIX
                        Restrict listing to parameters that have this prefix
  --source SOURCE       Source directory. Default: . (current dir)
  --build BUILD         Build directory. Default: build/
  -c, --clean           Clean the build directory before compiling
  -f, --flash           Flash the built firmware onto a connected target.
  --sterm               Open serial terminal after compiling. Can be chained
                        with --flash
  -N ARTIFACT_NAME, --artifact-name ARTIFACT_NAME
                        Name of the built program or library
  -S [{matrix,toolchains,targets}], --supported [{matrix,toolchains,targets}]
                        Shows supported matrix of targets and toolchains
  --app-config APP_CONFIG
                        Path of an application configuration file. Default is
                        to look for "mbed_app.json".
  -v, --verbose         Verbose diagnostic output
  -vv, --very_verbose   Very verbose diagnostic output

$ mbed --version
1.7.5

```
### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

kegilbert, 2018-09-05:
Looks like this might be a @theotherjimmy question.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-407*

closed 2019-05-22 by ciarmcom

0xc0170, 2019-05-22:
https://github.com/ARMmbed/mbed-os-5-docs/pull/820 fixed it