ARMmbed/mbed-os #7963: utf8 codec can't decode byte... when running Greentea

janjongboom, 2018-09-03:
### Description

I'm not sure if this is an issue with my target, or something else. But when running greentea I encounter the following:

```
mbedgt: test on hardware with target id: 46000000098be2c500000000000000000000000097969902
mbedgt: test suite 'mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler' ...... OK in 25.56 sec
        test case: 'Minimal Scheduler: Async Case 4 (Failure)' ....................................... OK in 0.24 sec
        test case: 'Minimal Scheduler: Case 1' ....................................................... OK in 0.02 sec
        test case: 'Minimal Scheduler: Case 2' ....................................................... OK in 0.01 sec
        test case: 'Minimal Scheduler: Case 3' ....................................................... OK in 0.02 sec
mbedgt: test case summary: 4 passes, 0 failures
mbedgt: mbed-host-test-runner: started
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Library/Python/2.7/site-packages/mbed_greentea-1.4.0-py2.7.egg/mbed_greentea/mbed_greentea_cli.py", line 508, in run_test_thread
    verbose=verbose)
  File "/Library/Python/2.7/site-packages/mbed_greentea-1.4.0-py2.7.egg/mbed_greentea/mbed_test_api.py", line 325, in run_host_test
    returncode, htrun_output = run_htrun(cmd, verbose)
  File "/Library/Python/2.7/site-packages/mbed_greentea-1.4.0-py2.7.egg/mbed_greentea/mbed_test_api.py", line 134, in run_htrun
    htrun_output += line.decode('utf-8')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe1 in position 33: invalid continuation byte
```

Not sure if this is a specific test, specific target, or something else. This was when running against RTL8170AF target on GCC ARM 6 on MacOS.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

janjongboom, 2018-09-03:
Quick fix (in `mbed_test_api.py`):

```python
htrun_output += line.decode('utf-8', 'ignore')
```

aashishc1988, 2018-09-11:
Wasn't this addressed in https://github.com/ARMmbed/greentea/pull/278 ?

cmonr, 2018-09-25:
Closing since it's been resolved.

closed 2018-09-25 by cmonr