ARMmbed/mbed-os #7972: Can not connect to second EthernetInterface

henningray, 2018-09-03:
### Description

Hello, I'm trying to create two EthernetInterfaces (see code below). The first one works fine, I can connect, print IP, and then disconnect. But for the second one the program gets stuck in connect(). I'm using Mbed OS 5.9.0 with GCC_ARM toolchain.

    NetworkInterface *net =new EthernetInterface();
    net->connect();
    const char *ip = net->get_ip_address();
    printf("IP address of device is: %s\n", ip ? ip : "No IP");
    net->disconnect();
    printf("Done\n");
    delete net;
    net = NULL;
    net = new EthernetInterface();
    printf("Connect 2nd time \n");
    net->connect();
    ip = net->get_ip_address();
    printf("IP address of device is: %s\n", ip ? ip : "No IP");
    net->disconnect();
    printf("Done\n");

The output is:

> IP address of device is: 192.168.2.3
> Done
> Connect 2nd time 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

kjbracey-arm, 2018-09-04:
This a known (but undocumented?) limitation - the plumbing between lwIP and the rest of the core is not capable of hot-**un**plugging interfaces, and never has been as far as I know. Once you've created an EthernetInterface and used it, you can't delete it and recreate it.

It should be possible to create two actually different interfaces and initialise both, but for your example there, you'll have to settle for not doing the `delete` and the `new` in the middle.

kjbracey-arm, 2018-09-04:
Basically it works because OnboardCellularInterface (as PPP) is effectively a singleton, and has a somewhat fake delete. When you delete the OnboardCellularInterface, you're not actually freeing up the singleton static backing data or or deleting lwIP's PPP interface - the second OnboardCellularInterface manages to reconnect to the same single PPP interface.

EthernetInterface is not a singleton - you can have multiple real EMAC-based interfaces, and they're created each time you make a new object. But the clean-up is not implemented.  We've not seen it as a priority to support removing network interfaces from stacks, as we expect device objects to usually be statically allocated so their memory is reserved up-front. I'm sure it's possible, but adds a whole new set of test coverage.

What's your use case, by the way?

loverdeg-ep, 2018-09-04:
Can't speak for @henningray but my use case would be to free up RAM if the interface isn't needed.

For example:
* A board that has both ethernet and wi-fi interfaces
* An application running on the board uses ethernet xor wi-fi

I suppose the safer solution might be to select the network interface at startup or manage it with multiple applications, but it might be nice to switch them out at run-time, post initialization.

kjbracey-arm, 2018-09-04:
Yes, I've had occasional discussions about this sort of "modal switch", and it seems reasonable to support it in principle.

My general caveat is that dynamic allocation for fundamental functionality can be potentially problematic, in terms of ensuring the memory is available and avoiding heap fragmentation. If memory is limited enough that you /need/ to do the modal switch and can't afford to have both interface stuff allocated at once, then there will likely be dangers of not being able to allocate the memory when you do the switch.

To avoid init problems, we do quite like to do static rather than dynamic allocation. I wonder if this could be done without a heap dependency via some sort of tricky placement new trickery akin to `SingletonPtr` - have static memory reserved big enough to placement new either interface one at a time.

I guess when you are dealing with a pair of Ethernet/Wifi interfaces, the memory in question can be quite significant - all those RX buffers. We're currently looking at paring those back separately - it seems most drivers are allocating far more RX buffers than they need, and generally not using them that efficiently.

henningray, 2018-09-05:
Main reason indeed would be to free up RAM that another part of the program (not allowed simultaneously with the connection of course). I agree that static allocation would be safer. However, if the application handles the memory well (freeing all memory it uses) it should be possible to use the same RAM space right? Or would you still have issues with memory fragmentation?
For now however, I guess that we’d be forced to create a global interface and just call the connect and disconnect functions accordingly, right?

kjbracey-arm, 2018-09-05:
If the application freed everything, fine, but if you're just using the general purpose heap, the typical problem is:

* Heap space 24K (say)
* Allocate 15K for device A (let's say it's a single contiguous buffer lump)
* Various parts of the system allocate various things, they're placed 15K into the heap
* Device A freed
* Device B needs 18K, but there's no contiguous 18K in the heap because of those allocations 15K into it, placed on top of where device A was.

The general purpose heaps in use by malloc and new can't really avoid this problem - and there are too many other people using the general heap - even if the application thinks it's freed everything bits of the OS or libraries may have left something there.

Some parts of the system use "nsdynmemlib" which has a simple heuristic that somewhat alleviates the problem by separating "temporary" and "permanent" allocations and bunches them at opposite ends of the heap. But that requires users that know to flag allocations, so doesn't help for plain malloc. And still doesn't really tackle this case, which is hard.

What I'm imagining is something that works like `SingletonPtr` where the space is reserved - basically have

     union MyInterfaces {
           EthernetInterface ethernet;
           SomeOtherInterface other;
     }

Then statically allocate `char interface_mem[sizeof MyInterfaces]`, and have something that does `new (interface_mem) EthernetInterface` to create it in that static memory position.

Not sure how you wrap that up into something neat and easy like `SingletonPtr`, but could do it manually.

If you could actually delete and new the interfaces, of course.

loverdeg-ep, 2018-09-05:
@kjbracey-arm 

I think I've heard, what you are referring to, called something like "static scratch space"?
"static overlay" or something similar?

kjbracey-arm, 2018-09-05:
I've sure someone has done something similar before.

Overlays were certainly more of a thing in the past - I remember very early ARM linkers that had an automatic code overlay mechanism. These constrained Mbed OS platforms could certainly benefit from some old-school concepts - things that have not been applicable to the desktop space for decades can be of interest again here. Explicit or toolchain-assisted overlays have not really been relevant for the desktop since VM paging.

This concept does also hinge on the theory that all the memory is actually embedded in the object, and it doesn't just have a bunch of `new` calls of its own inside its constructor, which would defeat the idea.

closed 2018-10-02 by adbridge