ARMmbed/mbed-os #7986: ISR Queue overflow that could potentially be avoided

marcemmers, 2018-09-04:
### Description

I have been getting some mbed errors that report an ISR queue overflow. 

After some digging i've found out that the source is an interrupt pin that is toggling a bit and is used to set a thread flag on both edges which calls `osRtxPostProcess` down the line. This causes the `os_isr_queue` to overflow so the error is valid.

I am aware of the fact that I should solve this in my application because this is putting a bigger load on the OS than necessary but still I wondered why this is happening. To me it seems rather silly that the entire buffer could be filled with the same `os_object_t` causing the hang up.

I have tested a simple solution that checks whether an `os_object_t` is already present in the queue before adding it. What I don't know if this would have a negative impact somewhere else in the OS or if some parts rely on multiple entries in the queue.

This was based on mbed version 5.9.3 but this part has not been changed since.

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

kjbracey-arm, 2018-09-05:
You have identified an issue I reported a while back, and you seem to have implemented one of the possibilities I suggested, but didn't actually try: https://github.com/ARM-software/CMSIS_5/issues/283

Fascinated to see that concept actually works - I wasn't sure if the "no duplicate" rule I hand-waved about would work in practice or there was some complication.

Your implementation could probably be streamlined by having a "queued for post-processing" flag in the `os_object_t` so you don't need to search. Although if we assume the queue is normally empty or nearly empty, maybe the search is no big deal.

> I am aware of the fact that I should solve this in my application because this is putting a bigger load on the OS than necessary

Well, if the OS was doing what you suggest, the load would be pretty minor - from ISR context you're not even going through an SVC call to reach the OS. Ultimately given the typical speed/space limits on Mbed OS platforms, I think it would be better to not require the duplicate suppression in every ISR handler (which is often a fiddle and prone to race conditions) but have it in the OS.

I would suggest you add a comment to that issue over on the CMSIS/RTX repo - this isn't really in-scope for mbed OS itself.

marcemmers, 2018-09-10:
I had searched the repository for duplicate issues but could not find one easily. In hindsight there is another issue reported here which had kind of the same issue but was solved on the application side; #6714.

My solution was just a proof of concept and could indeed be optimized by a flag of some sort. Even if the search could be fast I think its best to always minimize the time needed in such low level code.

I also did not know about the RTX repo, I will comment on your issue.

deepikabhavnani, 2018-09-18:
Duplicate of RTOS issue: https://github.com/ARM-software/CMSIS_5/issues/283

closed 2018-09-18 by deepikabhavnani

Yuri-Njathi, 2019-08-20:
++ MbedOS Error Info ++
Error Status: 0x80020126 Code: 294 Module: 2
Error Message: CMSIS-RTOS error: ISR Queue overflow
Location: 0x801502B
Error Value: 0x2
Current Thread: Id: 0x20000478 Entry: 0x80151FF StackSize: 0x200 StackMem: 0x20000508 SP: 0x2001FF98
-- MbedOS Error Info --

0xc0170, 2019-08-20:
@Yuri-Njathi We need more details to be able to review it. You can find some resolved issues  with the same error message and find out how to resolve it