ARMmbed/mbed-os #7997: Wake-up/sleep doesn't properly executed starting on its 2nd entry

gigz09, 2018-09-05:
### Description

I am currently working on an application that's going to be on-sleep whenever I want to and then wake-up on a specified period. On my test setup, I made an application that's going to toggle an LED after 5 seconds of sleep, at startup the pin goes high then goes low until the specified wake up period but starting on its 2nd loop the pin toggles on ~2.5mS while ST-Link is still connected if its disconnected it seems its not exiting sleep mode (pin only goes high). Sorry if this a dumb question but what could be wrong?

MCU: STM32L073RZ (custom board)
Toolchain: GNU ARM 6 2017-q2-update
IDE: Eclipse IDE
mbed-os version: 5.9.5 (f8b140f8d)

---

custom_targets.json
```javascript
{
    "Custom_L073RZ": {
        "inherits": ["FAMILY_STM32"],        
        "core": "Cortex-M0+",
        "extra_labels_add": ["STM32L0", "STM32L073RZ", "STM32L073xx"],
        "overrides": {
            "stdio_uart_tx": "PA_9",
            "stdio_uart_rx": "PA_10"
        },
        "config": {
            "clock_source": {
                "help": "Mask value : USE_PLL_HSE_EXTC | USE_PLL_HSE_XTAL (need HW patch) | USE_PLL_HSI ",
                "value": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
                "macro_name": "CLOCK_SOURCE"
            },
            "lpticker_lptim": {
                "help": "This target supports LPTIM. Set value 1 to use LPTIM for LOWPOWERTIMER, or 0 to use RTC wakeup timer",
                "value": 1
            }
        },
        "detect_code": ["0760"],
        "device_has_add": ["ANALOGOUT", "CRC", "SERIAL_FC", "SERIAL_ASYNCH", "TRNG", "FLASH"],
        "release_versions": ["2", "5"],
        "device_name": "STM32L073RZ"
    }
}
```

Here's my test code:
```c
#include "mbed.h"
#include "rtc_api_hal.h"

DigitalOut led(PA_0);

static RTC_HandleTypeDef RtcHandle;

void set_sleep(int32_t wakeup);

int main (void)
{
    if (!rtc_isenabled())
    {
        rtc_init();
        wait_us(250);

        HAL_NVIC_SetPriority(RTC_IRQn, 1, 0);
        HAL_NVIC_EnableIRQ(RTC_WKUP_IRQn);
    }

    while (1)
    {
        HAL_RTCEx_DeactivateWakeUpTimer(&RtcHandle);
        led = !led;
        set_sleep(5000); //wake up after 5seconds
    }
}

void set_sleep(int32_t wakeup)
{
    uint64_t wakeup_counter    = 0;
    uint32_t wakeup_clock 	= RTC_WAKEUPCLOCK_RTCCLK_DIV8; // 32768 / 8 = 4096 ticks per Second

    if (0 < wakeup)
    {
        core_util_critical_section_enter();

        // Set to Clear Wakeup Flag bit (WUTF)
        SET_BIT(PWR->CR, PWR_CR_CWUF);

        wakeup_counter  = (uint64_t)(wakeup * 4096);
        wakeup_counter /= 1000;
        wakeup_counter -= 1;

        set_time(0);
        RtcHandle.Instance = RTC;
        (void)HAL_RTCEx_SetWakeUpTimer_IT(&RtcHandle, (uint32_t)wakeup_counter, wakeup_clock);

        core_util_critical_section_exit();
    }
    
    sleep();
}
```

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

cmonr, 2018-09-06:
@gigz09 Could you clarify if this is with the ARM compiler or GCC ARM?

gigz09, 2018-09-06:
hi @cmonr, i believe its gcc arm https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads/6-2017-q2-update

jeromecoutant, 2018-09-06:
Hi
I don't understand why you are re-coding LPTICKER feature ?

gigz09, 2018-09-06:
hi @jeromecoutant,

this was actually coded first at v5.8.3 in which i believe doesn't support the lpticker yet. I've tried using the 'lp_ticker_set_interrupt' on v5.9.5 also but it doesn't behave the way I wanted to, I am already puzzled on what's going on so I stick to the method which gave me so far an output.

0xc0170, 2018-09-07:
> this was actually coded first at v5.8.3 in which i believe doesn't support the lpticker yet. I've tried using the 'lp_ticker_set_interrupt' on v5.9.5 also but it doesn't behave the way I wanted to, 

Apart from the above issue, what is the issue with 5.9.5 ? what are desired behavior and actual one?

gigz09, 2018-09-07:
hi @0xc0170,

Didn't experienced any other issue so far apart from what I have mentioned on this thread.

For now, I just want to make the sleep/wake-up work on the minimum of 100mS interval on a continuous loop. If 100mS is not achievable then anything near it would do.

LMESTM, 2018-09-07:
@gigz09 @0xc0170  @jeromecoutant @c1728p9 

The above example code calls for sleep() - which basically means WFI() only.  I _think_ that the target is not TICKLESS, so that means that the os ticker would immediately wake-up the target.  You may want to actually call deepsleep or consider the idea to move to a TICKLESS system which will cause the mbed-os background loop to call deepsleep for you. 

Note that what @gigz09 is looking for is an RTC alarm to minimize the number of wake-ups due to ticker wrap-around (even if the overhead of wake-up for ticker update is very low as already demonstrated by @c1728p9 in previous discussions). As commented by @jeromecoutant , the RTC is already being controled and used by HAL drivers, so it may not be a good idea to access to those registers from application. 

Maybe what @gigz09 need is an working example of a blinky on a TICKLESS target ?

gigz09, 2018-09-08:
hi @LMESTM,

I already did try calling 'hal_deepsleep' and calling stop mode through HAL but experienced the same.

I appreciate if someone could provide a working blinky on TICKLESS mode as guide.

jeromecoutant, 2018-09-10:
here is an example :

```
#include "mbed.h"
DigitalOut led1(LED1);
volatile uint32_t ticker_callback_count = 0;

void led_blink(void) {
        led1 = !led1;
        ticker_callback_count++;
}

int main()
{
    mbed_stats_cpu_t stats; // MBED_CPU_STATS_ENABLED or MBED_ALL_STATS_ENABLED
    LowPowerTicker mytimer;

#if MBED_TICKLESS
    printf("MBED_TICKLESS is enabled\n");
#else
    printf("MBED_TICKLESS is disabled\n");
#endif

    mytimer.attach(callback(led_blink), 1);

    while (1) {
        Thread::wait(10000);
        mbed_stats_cpu_get(&stats);
        printf("CPU %lu : up time %9llu (sleep %2llu%%, deepsleep %2llu%%)\n", ticker_callback_count, stats.uptime, (stats.sleep_time * 100) / stats.uptime, (stats.deep_sleep_time * 100) / stats.uptime);
    }
}

```

$ mbed compile -t GCC_ARM -m NUCLEO_F401RE -f --sterm -DMBED_ALL_STATS_ENABLED
...
MBED_TICKLESS is disabled
CPU 9 : up time  10026245 (sleep 98%, deepsleep  0%)
CPU 20 : up time  20079711 (sleep 98%, deepsleep  0%)
CPU 30 : up time  30134155 (sleep 98%, deepsleep  0%)
...
$ mbed compile -t GCC_ARM -m NUCLEO_F401RE -f --sterm -DMBED_ALL_STATS_ENABLED -DMBED_TICKLESS
...
MBED_TICKLESS is enabled
CPU 10 : up time  10025268 (sleep  0%, deepsleep 99%)
CPU 20 : up time  20080444 (sleep  0%, deepsleep 99%)
CPU 30 : up time  30135253 (sleep  0%, deepsleep 99%)
...

LMESTM, 2018-09-12:
@gigz09  have you seen the proposed example from @jeromecoutant  ?

gigz09, 2018-09-18:
hi @jeromecoutant and @LMESTM,

I tried this without the stats, the pin is toggling at two different rate, I'll capture a a snapshot when I got home. Any probability that my issue is on the config file?

chuanga, 2018-10-04:
@jeromecoutant I am investigating enabling lowpower feature for a customer using STM32L486 chipset and tried your example program above to enable MBED_TICKLESS. The program worked and it shows that it's at deep_sleep 99% of the time. 

However, it also seems that I can no longer update firmware via Daplink. Drag & drop to Daplink folder will always failed with FAIL.TXT ("error: The interface firmware FAILED to reset/halt the target MCU"). 

Have you encountered this issue? Thanks

jeromecoutant, 2018-10-04:
Hi

About daplink failure, no, I don't know this issue.

You should try to update your FW to the latest version:
https://www.st.com/en/development-tools/stsw-link007.html

About added note:

> added "LOWPOWERTIMER" to "device_has_add" section

This is not needed :-) as LOWPOWERTIMER macro has been replaced by LPTICKER

> removed ""device_has_removed": ["LPTICKER"]" section

This is correct. This is mandatory for TICKLESS.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-414*

chuanga, 2018-10-04:
@jeromecoutant  Thanks for the confirmation on the configuration. My board doesn't have ST-link debugger, only Daplink. I will try to get another board and see if it's H/W or S/W issue.

jeromecoutant, 2018-10-16:
Maybe we can close this issue ?
Thx

jeromecoutant, 2018-11-06:
ST_TO_BE_CLOSED