ARMmbed/mbed-os #8000: Nordic nRF52 BLE UARTService

wlamprecht, 2018-09-05:
### Description

Hi, 

I implemented the UART service as described in https://os.mbed.com/teams/Bluetooth-Low-Energy/wiki/UART-access-over-BLE. I can send data from the nrF Toolbox (Android) to the dev board and vice versa. But if I try to send more data than the max MTU size, the phone disconnects.  I sniffed the BLE traffic and everything seems ok>
[image]

But I do not receive any data. Did this work with other boards? 

Tagret: Nordic nRF52
Toolchain: mbed cli 1.7.5, arm-gcc

Thanks, 
Werner 

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPAN-261*

desmond-blue, 2019-01-03:
This is related to #8548

wlamprecht, 2019-01-15:
Yes

closed 2019-01-15 by wlamprecht