ARMmbed/mbed-os #8004: ARMC6: legacy align error

0xc0170, 2018-09-05:
### Description

This PR removed legacy align : https://github.com/ARMmbed/mbed-os/pull/7792 . It produces warning. This removed warning but causing now failures to use scatter files we have in the code base as they rely on the alignment that is now legacy. This leads to linker errors (4 vs 8 byte alignment).  I could not find documentation why this alignment was deprecated but causing "breaking change" in this case as we would need to update all scatter files to use a new alignment. 

@deepikabhavnani @JanneKiiskila 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

deepikabhavnani, 2018-09-05:
As per the http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0803g/pge1362075504330.html, `--legacyalign, --no_legacyalign` are deprecated, hence were removed from mbed-os.

We have two options to resolve the issue:
1. Fix the sections in scatter file to fix the alignment issue
2. Enabling deprecated option --legacyalign

I would prefer option 1.

cmonr, 2018-09-06:
Going to keep this open for now even though https://github.com/ARMmbed/mbed-os/pull/8012 has been merged.
If we want to remove the legacyalign flag, we should fix the files and figure out why the deperecated flag led to behavior changes.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-415*

PR #8186 (2018-10-10): Freescale/NXP: Fix alignment of execute region to 8byte boundary 
### Description

Resolves: #8004

### Pull request type
    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-10 by cmonr