ARMmbed/mbed-os #8007: STM32L053C8 Reduced Flash Size on Win10

Tony-YI, 2018-09-05:
### Description
Currently I am working on a custom board utilizing the STM32L053C8 and I am programming it using the ST-Link on a NUCLEO board. 

I noticed that the flash size shown on Win10 is 52KB instead of 64KB. However, after I tried to format the disk (which can not be actually performed), the flash size displayed on Win10 became 64KB and I could program the MCU then because my program is around 63KB.

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

bcostm, 2018-09-06:
Can you update to the latest STLink firmware and try again ?
https://os.mbed.com/teams/ST/wiki/Nucleo-Firmware

jeromecoutant, 2018-09-06:
@schstm

Tony-YI, 2018-09-06:
@bcostm 

It work! Thank you all very much for the prompt reply.

closed 2018-09-06 by Tony-YI