ARMmbed/mbed-os #8011: mbed_drivers-crc tests fails on EFM32PG_STK3401

studavekar, 2018-09-05:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
EFM32PG_STK3401

**Toolchain:**
GCC_ARM

**mbed-os sha:**

06106297a (HEAD -> master, origin/master, origin/HEAD) Merge pull request #6293 from Taiki-San/patch-

**Steps to reproduce**

Use command run :  mbed test -m EFM32PG_STK3401 -t GCC_ARM -n tests-mbed_drivers-crc -c -v

logs snippet
```
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m EFM32PG_STK3401 -p /dev/ttyACM34:115200 -f "BUILD/tests/EFM32PG_STK3401/GCC_ARM/TESTS/mbed_drivers/crc/crc.bin" -e "TESTS/host_tests" -d /mnt/pci-0000_00_14_0-usb-0_7_2_7_1_3-scsi-0_0_0_0 -c default -t 2035022D000122D7D4751138 -r default -C 4 -R 2 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1536179180.54][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
[1536179180.55][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
[1536179180.56][HTST][INF] host test executor ver. 1.4.1
[1536179180.56][HTST][INF] copy image onto target...
[1536179180.56][COPY][INF] Waiting up to 60 sec for '2035022D000122D7D4751138' mount point (current is '/mnt/pci-0000_00_14_0-usb-0_7_2_7_1_3-scsi-0_0_0_0')...
[1536179191.07][HTST][INF] starting host test process...
[1536179191.07][CONN][INF] starting connection process...
[1536179191.07][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1536179191.07][CONN][INF] initializing serial port listener...
[1536179191.07][PLGN][INF] Waiting up to 60 sec for '2035022D000122D7D4751138' serial port (current is '/dev/ttyACM34')...
[1536179191.07][HTST][INF] setting timeout to: 60 sec
[1536179191.35][mbedls.lstools_base]No valid file found to update JLink device details
[1536179191.35][mbedls.lstools_base]No valid file found to update JLink device details
[1536179191.35][mbedls.lstools_base]daplink entry: "0000" not found in platform database
[1536179191.35][mbedls.lstools_base]No valid file found to update JLink device details
[1536179191.35][mbedls.lstools_base]daplink entry: "7012" not found in platform database
[1536179191.36][SERI][INF] serial(port=/dev/ttyACM34, baudrate=115200, read_timeout=0.01, write_timeout=5)
[1536179191.36][SERI][INF] reset device using 'default' plugin...
[1536179191.46][SERI][INF] waiting 2.00 sec after reset
[1536179193.46][SERI][INF] wait for it...
[1536179193.47][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1536179193.47][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1536179193.47][CONN][INF] sending preamble 'a127bea3-e544-40ee-9e19-81487c316f0a'
[1536179193.49][SERI][TXD] {{__sync;a127bea3-e544-40ee-9e19-81487c316f0a}}
[1536179193.51][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1536179193.51][CONN][INF] found SYNC in stream: {{__sync;a127bea3-e544-40ee-9e19-81487c316f0a}} it is #0 sent, queued...
[1536179193.51][HTST][INF] sync KV found, uuid=a127bea3-e544-40ee-9e19-81487c316f0a, timestamp=1536179193.508620
[1536179193.51][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1536179193.51][CONN][INF] found KV pair in stream: {{__timeout;15}}, queued...
[1536179193.51][HTST][INF] DUT greentea-client version: 1.3.0
[1536179193.51][HTST][INF] setting timeout to: 15 sec
[1536179193.52][CONN][RXD] >>> Running 5 test cases...
[1536179193.52][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1536179193.52][CONN][INF] found KV pair in stream: {{__testcase_count;5}}, queued...
[1536179193.52][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1536179193.52][HTST][INF] host test setup() call...
[1536179193.52][HTST][INF] CALLBACKs updated
[1536179193.52][HTST][INF] host test detected: default_auto
[1536179193.53][CONN][INF] found KV pair in stream: {{__testcase_name;Test supported polynomials}}, queued...
[1536179193.53][CONN][INF] found KV pair in stream: {{__testcase_name;Test partial CRC}}, queued...
[1536179193.53][CONN][INF] found KV pair in stream: {{__testcase_name;Test SD CRC polynomials}}, queued...
[1536179193.55][CONN][RXD]
[1536179193.55][CONN][RXD] >>> Running case #1: 'Test supported polynomials'...
[1536179193.55][CONN][INF] found KV pair in stream: {{__testcase_name;Test not supported polynomials}}, queued...
[... 41 lines cut ...]
        reported by utest: passes = 0, failures 1)
        test case result count: passes = 0, failures 5)
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.4420808959
mbedgt: test suite report:
+-------------------------+-----------------+------------------------+--------+--------------------+-------------+
| target                  | platform_name   | test suite             | result | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+------------------------+--------+--------------------+-------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-crc | FAIL   | 13.3               | default     |
+-------------------------+-----------------+------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 FAIL
mbedgt: test case report:
+-------------------------+-----------------+------------------------+--------------------------------+--------+--------+---------+--------------------+
| target                  | platform_name   | test suite             | test case                      | passed | failed | result  | elapsed_time (sec) |
+-------------------------+-----------------+------------------------+--------------------------------+--------+--------+---------+--------------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-crc | Test SD CRC polynomials        | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-crc | Test not supported polynomials | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-crc | Test partial CRC               | 0      | 0      | SKIPPED | 0.0                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-crc | Test supported polynomials     | 0      | 1      | FAIL    | 0.0                |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-crc | Test thread safety             | 0      | 0      | SKIPPED | 0.0                |
+-------------------------+-----------------+------------------------+--------------------------------+--------+--------+---------+--------------------+

```

CC : @maclobdell @bentcooke

0xc0170, 2018-09-06:
cc @ARMmbed/team-silabs

stevew817, 2018-09-06:
Yikes, looks like the issue is on our end. I'll submit a PR asap.

kjbracey-arm, 2018-09-24:
Is this now resolved?