ARMmbed/mbed-os #8021: STM32: gpio_irq_disable Disables all InterruptIn Instances Sharing EXTI9_5 or EXTI15_10

mattbrown015, 2018-09-06:
### Description
I've just discovered that using `InterruptIn`  on STM32 devices is more complicated than I thought.

Unfortunately our board has digital inputs on PA7 and PB5 and when I disable PA7 PB5 is also disabled.

It seems to me that `gpio_irq_disable` is too simplistic in the way it disables the IRQ without considering if it is being used by another input.

Unfortunately I can see that handling the shared interrupt vectors in a way that is completely consistent and predictable is a problem.

The fact that calling `InterruptIn::disable_irq()` may actually disable multiple `InterruptIn` instances feels like a bug to me.  Although the STM team may disagree!

I shall have a rummage around in `targets\TARGET_STM\gpio_irq_api.c` to see if there is an easy solution to my problem.  In the mean time I wondered if the STM team had any thoughts.

This might be related to [TARGET_STM/gpio_irq_api.c does not need to guess which port generated the interrupt #6722](https://github.com/ARMmbed/mbed-os/issues/6722).
### Issue request type
[ ] Question
[ ] Enhancement
[X] Bug

mattbrown015, 2018-09-07:
I've got a suggestion!

We can use the channel pin mask to check how many pins are currently being used to generate IRQs on this channel/vector.

Here's what I've just tried.  It solves my problem but I haven't tried enabling and disabling IRQs on all combinations of pins and vectors.

    diff --git a/targets/TARGET_STM/gpio_irq_api.c b/targets/TARGET_STM/gpio_irq_api.c
    index b3d1cd200..63ee2a3dc 100644
    --- a/targets/TARGET_STM/gpio_irq_api.c
    +++ b/targets/TARGET_STM/gpio_irq_api.c
    @@ -27,7 +27,7 @@
      * OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
      *******************************************************************************
      */
    -#include <stddef.h>
    +#include <stdbool.h>
     #include "cmsis.h"
     #include "gpio_irq_api.h"
     #include "pinmap.h"
    @@ -268,11 +268,11 @@ void gpio_irq_free(gpio_irq_t *obj)
         uint32_t gpio_idx = pin_lines_desc[STM_PIN(obj->pin)].gpio_idx;
         gpio_channel_t *gpio_channel = &channels[obj->irq_index];
     
    -    gpio_irq_disable(obj);
         gpio_channel->pin_mask &= ~(1 << gpio_idx);
         gpio_channel->channel_ids[gpio_idx] = 0;
         gpio_channel->channel_gpio[gpio_idx] = 0;
         gpio_channel->channel_pin[gpio_idx] = 0;
    +    gpio_irq_disable(obj);
     }
     
     void gpio_irq_set(gpio_irq_t *obj, gpio_irq_event event, uint32_t enable)
    @@ -325,10 +325,20 @@ void gpio_irq_enable(gpio_irq_t *obj)
     
     void gpio_irq_disable(gpio_irq_t *obj)
     {
    +    const uint32_t pin_index = STM_PIN(obj->pin);
    +    const uint32_t gpio_idx = pin_lines_desc[pin_index].gpio_idx;
    +    const uint32_t pin_mask = 1 << gpio_idx;
    +    const uint32_t irq_index = pin_lines_desc[pin_index].irq_index;
    +    const gpio_channel_t *const gpio_channel = &channels[irq_index];
    +
         /* Clear EXTI line configuration */
    -    LL_EXTI_DisableRisingTrig_0_31(1 << STM_PIN(obj->pin));
    -    LL_EXTI_DisableFallingTrig_0_31(1 << STM_PIN(obj->pin));
    -    LL_EXTI_DisableIT_0_31(1 << STM_PIN(obj->pin));
    -    NVIC_DisableIRQ(obj->irq_n);
    -    NVIC_ClearPendingIRQ(obj->irq_n);
    +    LL_EXTI_DisableRisingTrig_0_31(1 << pin_index);
    +    LL_EXTI_DisableFallingTrig_0_31(1 << pin_index);
    +    LL_EXTI_DisableIT_0_31(1 << pin_index);
    +
    +    const bool no_more_pins_on_vector = (gpio_channel->pin_mask & ~pin_mask) == 0;
    +    if (no_more_pins_on_vector) {
    +        NVIC_DisableIRQ(obj->irq_n);
    +        NVIC_ClearPendingIRQ(obj->irq_n);
    +    }
     }

LMESTM, 2018-09-07:
@mattbrown015 thanks for sharing a patch ! 
Looks good at first glance. would you mind sending a PR and @jeromecoutant we can run non regression testing on the PR branch

mattbrown015, 2018-10-03:
I believe this was fixed by [STM32: Fix disabling of IRQs shared by multiple events #8037](https://github.com/ARMmbed/mbed-os/pull/8037).

Closing...

closed 2018-10-03 by mattbrown015