ARMmbed/mbed-os #8027: Support SKT LoRaWAN variant

janjongboom, 2018-09-07:
### Description

I'm not sure if we should support it, but SKT in Korea uses a non-standard version of LoRaWAN. It first does a normal join, then exchanges `AppKeyAllocReq`, `AppKeyAllocAns`, `AppKeyReportReq` and `AppKeyReportAns` commands on port 223 and then re-joins. The Redwood RWC5020A can emulate these. Documentation (in Korean) is here: 
https://www.sktiot.com/api/common/file/download?fileId=00EU2GOQPM1E9FDBDAAA

Some things to consider:

* Should we do a reference application which implements it, or implement in stack?
* Should we even support non-standard extensions?

SKT has nation-wide coverage in Korea, so would be a good opportunity. Apparently there's a reference implementation on top of Murata so we could look at that.

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

janjongboom, 2018-09-07:
Here's a demo program which uses the SKT join procedure (verified on RWC5020A): https://github.com/janjongboom/mbed-os-example-lorawan-skt

AnttiKauppila, 2018-09-11:
At the moment this will not be supported. We are only focusing to standard implementation now.

closed 2018-10-02 by adbridge