ARMmbed/mbed-os #8038: UpdSocket limitations

jagomo, 2018-09-07:
### Description

## Context

**Target** NUCLEOF412ZG + Atmel AT86RF233
**Toolchain** GCC_ARM
**mbed-os SHA** 62f8b922b420626514fd4690107aff4188469833
**mbed-cli** 1.7.5

I'm trying to build an application where I need to broadcast to some wireless Thread nodes small pieces of data (<16bytes). I used as template the mbed-os-example-mesh-minimal. 

To try if our requirements are met, I modified slightly the example. Now, instead of requiring a push of a button to broadcast a message, one of the boards does it periodically. I've observed that if we go for small periods (5ms) at some point the sending will start to fail. First sometimes, then more often and finally crashing the device

```
<\r><27>[2K<27>[90m[DBG ][nsif]: socket_sendto(socket=0x200065f4) sock_id=4, ret=11<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][buff]: buffer_get failed: alloc(zd)<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][mpl ]: seed 38:05 seq 34<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][iphc]: No mem<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][nsif]: socket_sendto(socket=0x200065f4) sock_id=4, ret=11<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][buff]: buffer_get failed: alloc(zd)<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][mpl ]: seed 38:05 seq 35<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][iphc]: No mem<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][nsif]: socket_sendto(socket=0x200065f4) sock_id=4, ret=11<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][buff]: buffer_get failed: alloc(zd)<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][mpl ]: seed 38:05 seq 36<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][iphc]: No mem<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][nsif]: socket_sendto(socket=0x200065f4) sock_id=4, ret=11<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][buff]: buffer_get failed: alloc(zd)<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][mpl ]: seed 38:05 seq 37<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][iphc]: No mem<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][nsif]: socket_sendto(socket=0x200065f4) sock_id=4, ret=11<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][mPDs]: pd_ind buffer get fail 68<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][mMCp]: MAC tx fail<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][thrd]: In Thread TX_FAIL handler, accumulated_failures=16<27>[0m<\r><\n>
<\r><27>[2K<27>[33m[WARN][thrd]: End device lost parent, reset!<\r><\n>
<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][ThSA]: Service 7, Uri unregistration uri: a/aq<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][ThSA]: Service 8, Uri unregistration uri: a/an<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][MACh]: unRegister Device<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][mMIB]: Remove user 0 from key<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][mMIB]: Over write 0, mac16 3800 mac64: 8a:e8:94:cc:e9:90:89:25, 45981<27>[0m<\r><\n>
<\r><27>[2K<27>[90m[DBG ][mMIB]: Set 0, mac16 ffff mac64: ff:ff:ff:ff:ff:ff:ff:ff, 4294967295<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 233<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><27>[2K<27>[39m[INFO][6lAd]: Dest addr: 3800<27>[0m<\r><\n>
<\r><27>[2K<27>[31m[ERR ][6lAd]: MCPS Data fail by status 243<27>[0m<\r><\n>
<\r><\n>
++ MbedOS Fault Handler ++<\r><\n>
<\r><\n>
FaultType: HardFault<\r><\n>
<\r><\n>
Context:<\r><\n>
R0   : 20001704<\r><\n>
R1   : 00000000<\r><\n>
R2   : 00000018<\r><\n>
R3   : 534618B4<\r><\n>
R4   : 20001704<\r><\n>
R5   : 2000AE48<\r><\n>
R6   : 000003CC<\r><\n>
R7   : 00000005<\r><\n>
R8   : 00000002<\r><\n>
R9   : 0807AD7F<\r><\n>
R10  : 0807ADCB<\r><\n>
R11  : 00000001<\r><\n>
R12  : 080570D9<\r><\n>
SP   : 2003FFC0<\r><\n>
LR   : 0805713B<\r><\n>
PC   : 08057196<\r><\n>
xPSR : 210F000B<\r><\n>
PSP  : 20009740<\r><\n>
MSP  : 2003FFA0<\r><\n>
CPUID: 410FC241<\r><\n>
HFSR : 40000000<\r><\n>
MMFSR: 00000000<\r><\n>
BFSR : 00000082<\r><\n>
UFSR : 00000000<\r><\n>
DFSR : 0000000B<\r><\n>
AFSR : 00000000<\r><\n>
BFAR : 534618BC<\r><\n>
Mode : Handler<\r><\n>
Priv : Privileged<\r><\n>
Stack: MSP<\r><\n>
<\r><\n>
-- MbedOS Fault Handler --<\r><\n>
<\r><\n>
<\r><\n>
<\r><\n>
++ MbedOS Error Info ++<\r><\n>
Error Status: 0x80FF013D Code: 317 Module: 255<\r><\n>
Error Message: Unrecoverable fault excaption.<\r><\n>
Location: 0x805473B<\r><\n>
Error Value: 0x8057196<\r><\n>
Current Thread: Id: 0x2000AE48 Entry: 0x8021E09 StackSize: 0x1800 StackMem: 0x20009648 SP: 0x2003FF30 <\r><\n>
-- MbedOS Error Info --<\r><\n>
```
The final app will be interrupt driven, and several in short succession might happen in a window <8ms, 
If I simulate bursts by calling consecutively:
```
 my_socket->sendto(send_sockAddr, testbuffer, 11);
```
This problem gets worse.

Finally if I drive the sending with an interrupt, I can keep things stable as long as the interrupt is triggered with a period >60ms. Everything below that will generate the issue described. 

With bursts of interrupts triggered on and a 60 ms basis. No packet will be sent out at all until the device crashes. Then some of them will reach the rest of the boards in the mesh.

## Question

Is there any know limitations in UpdSocket? The API leads to understand that "sendto" is a blocking call, and therefore, being called consecutively should not matter as long it is done from the same context. 

At the same time, based in the experiments performed, looks like if an ongoing "sendto" operation is interrupted and a new one is issued, both will fail and the device will eventually crash. Is it possible to capture the SOCKET_TX_DONE event from the main application? this way I could try to bypass the current problem by blocking further calls until the transmission is done.

### Issue request type

[X] Question  
[ ] Enhancement  
[X] Bug?

jagomo, 2018-09-10:
I've just realized that on the mbed-os-example-mesh-minimal the socket is set to non-blocking. I guess that it relies on the fact that no one is physically able to press the button that triggers the broadcast fast enough. I guess then that therefore my test reach the breaking point of that concept.

So I guess we can focus more on the second question. Checking in more detail the socket documentation I found that the socket classes provide a sigio() member function to register a callback on socket state changes. Nevertheless the function registered does not feature input parameters. So how can we derive which event triggered the callback? (also is suggested that "The callback may also be called spuriously without reason")

SeppoTakalo, 2018-09-19:
Hi @jagomo 

The hard fault handler needs to be studied, are you able to attach a GDB or other debugger and provide a stack trace where the crash happened.

The test application itself is a bit unrealistic. You cannot expect that low power 802.15.4 network is able to send traffic in 5ms interval. Basically you are flooding the network, heap and event loop of the system.

Also, you mentioned that you are sending from interrupt, which should not be possible. Socket API should not be called from interrupt, because it uses mutexes and may allocate memory.

I assume that the crash might be related to exhausted memory from event system, or crash related to mutex handling in event loop or Socket API.

jagomo, 2018-09-19:
Hi @SeppoTakalo,

Thanks for the clarification

> Also, you mentioned that you are sending from interrupt, which should not be possible. Socket API should not be called from interrupt, because it uses mutexes and may allocate memory.

Just also to clarify when I wrote "Finally if I drive the sending with an interrupt..." I meant that instead of calling periodically every 5ms within a loop, an interrupt queues on an eventQueue the sending. So the sending does not happen in interrupt context. I wanted to give that piece of information because What I found weird is that with this method (triggering the scheduling by an interrupt into an event queue) the period times I can use reliably are >10x bigger (5ms vs 60 ms, before packets start being lost)

> The hard fault handler needs to be studied, are you able to attach a GDB or other debugger and provide a stack trace where the crash happened.

Next time I observe the issue I'll try to provide that.

For the time being and after, more test were we find out other limitations, we are going for a different approach as, as you point out, the expectations we have about the period might be way to high, but we wanted to see were we could get :)

I'll close this issue and reopen it in case of more questions revolving the socket limitations or if I can provide new information. Thank you for your time.

closed 2018-09-19 by jagomo

SeppoTakalo, 2018-09-19:
@jagomo Thanks for the info.

Also, you might want to study similar issue https://github.com/ARMmbed/mbed-os/issues/7204 where 1 Mhz packet interval was expected.

And if you can, I would like to have the test application if we can recreate the same test setup and study the hard fault.

jagomo, 2018-10-19:
@SeppoTakalo First of all sorry for the BIG delay. Answering this slipped through my TODO list. I just realized this was pending for answering.

Unfortunately we don't have the test code sitting around any more. I guess a simple and easy way of generating a similar behaviour would be to make a Ticker trigger the Button press callback on one of the boards with a very high frequency (5ms or so).