ARMmbed/mbed-os #8041: lp_ticker -Test multi ticker is fails on NUMAKER_PFM_M2351

studavekar, 2018-09-07:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_M2351

**Toolchain:**
ARMC6 

**mbed-os sha:**

6827c370 (HEAD -> master, origin/master, origin/feature-hal-spec-spi, origin/HEAD) Merge pull request #7971 from JammuKekkonen/fix_softdevice_memory_reservation_for_nrf52dk

**Steps to reproduce**

Use command run :  mbed test -m NUMAKER_PFM_M2351 -t ARMC6 -v -n tests-mbed_drivers-lp_ticker  -v

logs snippet
```
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUMAKER_PFM_M2351 -p /dev/ttyACM22:9600 -f "BUILD/tests/NUMAKER_PFM_M2351/ARMC6/TESTS/mbed_drivers/lp_ticker/lp_ticker.hex" -e "TESTS/host_tests" -d /mnt/pci-0000_00_14_0-usb-0_7_2_3_1_0-scsi-0_0_0_0 -c default -t 13050000002D0101013B97B000000002 -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1536339974.00][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
[1536339974.01][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
[1536339974.02][HTST][INF] host test executor ver. 1.4.1
[1536339974.02][HTST][INF] copy image onto target...
[1536339974.02][COPY][INF] Waiting up to 60 sec for '13050000002D0101013B97B000000002' mount point (current is '/mnt/pci-0000_00_14_0-usb-0_7_2_3_1_0-scsi-0_0_0_0')...
[1536339986.33][mbedls.lstools_base]No valid file found to update JLink device details
[1536339986.33][mbedls.lstools_base]daplink entry: "7012" not found in platform database
[1536339986.84][HTST][INF] starting host test process...
[1536339986.84][CONN][INF] starting connection process...
[1536339986.84][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1536339986.84][CONN][INF] initializing serial port listener...
[1536339986.84][PLGN][INF] Waiting up to 60 sec for '13050000002D0101013B97B000000002' serial port (current is '/dev/ttyACM22')...
[1536339986.84][HTST][INF] setting timeout to: 60 sec
[1536339987.12][mbedls.lstools_base]No valid file found to update JLink device details
[1536339987.12][mbedls.lstools_base]No valid file found to update JLink device details
[1536339987.12][mbedls.lstools_base]daplink entry: "0000" not found in platform database
[1536339987.12][mbedls.lstools_base]No valid file found to update JLink device details
[1536339987.12][mbedls.lstools_base]daplink entry: "7012" not found in platform database
[1536339987.12][SERI][INF] serial(port=/dev/ttyACM22, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1536339987.12][SERI][INF] reset device using 'default' plugin...
[1536339987.23][SERI][INF] waiting 1.00 sec after reset
[1536339988.23][SERI][INF] wait for it...
[1536339988.23][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1536339988.23][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1536339988.24][CONN][INF] sending preamble '5712f9a1-3011-467b-bdf5-d7dbc7f0fc23'
[1536339988.24][SERI][TXD] {{__sync;5712f9a1-3011-467b-bdf5-d7dbc7f0fc23}}
[1536339988.37][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1536339988.42][CONN][INF] found SYNC in stream: {{__sync;5712f9a1-3011-467b-bdf5-d7dbc7f0fc23}} it is #0 sent, queued...
[1536339988.43][HTST][INF] sync KV found, uuid=5712f9a1-3011-467b-bdf5-d7dbc7f0fc23, timestamp=1536339988.424627
[1536339988.44][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1536339988.44][HTST][INF] DUT greentea-client version: 1.3.0
[1536339988.47][CONN][INF] found KV pair in stream: {{__timeout;30}}, queued...
[1536339988.47][HTST][INF] setting timeout to: 30 sec
[1536339988.51][CONN][INF] found KV pair in stream: {{__host_test_name;timing_drift_auto}}, queued...
[1536339988.51][HTST][INF] host test class: '<class 'timing_drift_auto.TimingDriftSync'>'
[1536339988.51][HTST][INF] host test setup() call...
[1536339988.51][HTST][INF] CALLBACKs updated
[1536339988.51][HTST][INF] host test detected: timing_drift_auto
[1536339988.53][CONN][INF] found KV pair in stream: {{__testcase_count;11}}, queued...
[1536339988.55][CONN][RXD] >>> Running 11 test cases...
[1536339988.62][CONN][INF] found KV pair in stream: {{__testcase_name;Test attach for 0.001s and time measure}}, queued...
[1536339988.69][CONN][INF] found KV pair in stream: {{__testcase_name;Test attach_us for 1ms and time measure}}, queued...
[1536339988.75][CONN][INF] found KV pair in stream: {{__testcase_name;Test attach for 0.01s and time measure}}, queued...
[1536339988.81][CONN][INF] found KV pair in stream: {{__testcase_name;Test attach_us for 10ms and time measure}}, queued...
[... 95 lines cut ...]
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | TIMEOUT | 44.91              | default     |
+-------------------------+-------------------+------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
+-------------------------+-------------------+------------------------------+-------------------------------------------+--------+--------+--------+--------------------+
| target                  | platform_name     | test suite                   | test case                                 | passed | failed | result | elapsed_time (sec) |
+-------------------------+-------------------+------------------------------+-------------------------------------------+--------+--------+--------+--------------------+
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach for 0.001s and time measure   | 1      | 0      | OK     | 0.09               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach for 0.01s and time measure    | 1      | 0      | OK     | 0.07               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach for 0.1s and time measure     | 1      | 0      | OK     | 0.16               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach for 0.5s and time measure     | 1      | 0      | OK     | 0.56               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach_us for 100ms and time measure | 1      | 0      | OK     | 0.16               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach_us for 10ms and time measure  | 1      | 0      | OK     | 0.09               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach_us for 1ms and time measure   | 1      | 0      | OK     | 0.06               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test attach_us for 500ms and time measure | 1      | 0      | OK     | 0.55               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test detach                               | 1      | 0      | OK     | 0.72               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test multi call and time measure          | 1      | 0      | OK     | 1.06               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_ticker | Test multi ticker                         | 0      | 0      | ERROR  | 0.0                |
+-------------------------+-------------------+------------------------------+-------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 10 OK / 1 ERROR
mbedgt: completed in 45.30 sec
mbedgt: exited with code 
```

CC : @maclobdell @bentcooke @ccli8

ccli8, 2018-09-10:
#8029 would fix the failure.

0xc0170, 2018-09-19:
The fix was integrated, will be in 5.10.1

closed 2018-09-19 by 0xc0170