ARMmbed/mbed-os #8042: mbed_drivers-lp_timeout test  fails on NUMAKER_PFM_M2351

studavekar, 2018-09-07:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_M2351

**Toolchain:**
ARMC6 

**mbed-os sha:**

6827c370 (HEAD -> master, origin/master, origin/feature-hal-spec-spi, origin/HEAD) Merge pull request #7971 from JammuKekkonen/fix_softdevice_memory_reservation_for_nrf52dk

**Steps to reproduce**

Use command run :  mbed test -m NUMAKER_PFM_M2351 -t ARMC6  -n tests-mbed_drivers-lp_timeout -v

logs snippet
```
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUMAKER_PFM_M2351 -p /dev/ttyACM22:9600 -f "BUILD/tests/NUMAKER_PFM_M2351/ARMC6/TESTS/mbed_drivers/lp_timeout/lp_timeout.hex" -e "TESTS/host_tests" -d /mnt/pci-0000_00_14_0-usb-0_7_2_3_1_0-scsi-0_0_0_0 -c default -t 13050000002D0101013B97B000000002 -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1536341892.94][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
[1536341892.96][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
[1536341892.96][HTST][INF] host test executor ver. 1.4.1
[1536341892.96][HTST][INF] copy image onto target...
[1536341892.96][COPY][INF] Waiting up to 60 sec for '13050000002D0101013B97B000000002' mount point (current is '/mnt/pci-0000_00_14_0-usb-0_7_2_3_1_0-scsi-0_0_0_0')...
[1536341893.24][mbedls.lstools_base]No valid file found to update JLink device details
[1536341905.92][HTST][INF] starting host test process...
[1536341905.92][CONN][INF] starting connection process...
[1536341905.92][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1536341905.92][CONN][INF] initializing serial port listener...
[1536341905.92][PLGN][INF] Waiting up to 60 sec for '13050000002D0101013B97B000000002' serial port (current is '/dev/ttyACM22')...
[1536341905.92][HTST][INF] setting timeout to: 60 sec
[1536341906.21][mbedls.lstools_base]No valid file found to update JLink device details
[1536341906.21][mbedls.lstools_base]No valid file found to update JLink device details
[1536341906.21][mbedls.lstools_base]daplink entry: "0000" not found in platform database
[1536341906.21][mbedls.lstools_base]No valid file found to update JLink device details
[1536341906.22][mbedls.lstools_base]daplink entry: "7012" not found in platform database
[1536341906.22][SERI][INF] serial(port=/dev/ttyACM22, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1536341906.22][SERI][INF] reset device using 'default' plugin...
[1536341906.33][SERI][INF] waiting 1.00 sec after reset
[1536341907.33][SERI][INF] wait for it...
[1536341907.33][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1536341907.33][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1536341907.33][CONN][INF] sending preamble '455b3d7f-6c7d-4969-bc9d-16f827d70289'
[1536341907.33][SERI][TXD] {{__sync;455b3d7f-6c7d-4969-bc9d-16f827d70289}}
[1536341907.47][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1536341907.52][CONN][INF] found SYNC in stream: {{__sync;455b3d7f-6c7d-4969-bc9d-16f827d70289}} it is #0 sent, queued...
[1536341907.52][HTST][INF] sync KV found, uuid=455b3d7f-6c7d-4969-bc9d-16f827d70289, timestamp=1536341907.523959
[1536341907.53][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1536341907.54][HTST][INF] DUT greentea-client version: 1.3.0
[1536341907.57][CONN][INF] found KV pair in stream: {{__timeout;240}}, queued...
[1536341907.57][HTST][INF] setting timeout to: 240 sec
[1536341907.61][CONN][INF] found KV pair in stream: {{__host_test_name;timing_drift_auto}}, queued...
[1536341907.61][HTST][INF] host test class: '<class 'timing_drift_auto.TimingDriftSync'>'
[1536341907.61][HTST][INF] host test setup() call...
[1536341907.61][HTST][INF] CALLBACKs updated
[1536341907.61][HTST][INF] host test detected: timing_drift_auto
[1536341907.63][CONN][INF] found KV pair in stream: {{__testcase_count;22}}, queued...
[1536341907.66][CONN][RXD] >>> Running 22 test cases...
[1536341907.71][CONN][INF] found KV pair in stream: {{__testcase_name;Callback called once (attach)}}, queued...
[1536341907.77][CONN][INF] found KV pair in stream: {{__testcase_name;Callback called once (attach_us)}}, queued...
[1536341907.84][CONN][INF] found KV pair in stream: {{__testcase_name;Callback not called when cancelled (attach)}}, queued...
[1536341907.90][CONN][INF] found KV pair in stream: {{__testcase_name;Callback not called when cancelled (attach_us)}}, queued...
[1536341907.95][CONN][INF] found KV pair in stream: {{__testcase_name;Callback override (attach)}}, queued...
[... 112 lines cut ...]
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | 1 s delay during sleep (attach)                   | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | 1 s delay during sleep (attach_us)                | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | 10 ms delay accuracy (attach)                     | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | 10 ms delay accuracy (attach_us)                  | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | 5 s delay accuracy (attach)                       | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | 5 s delay accuracy (attach_us)                    | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Callback called once (attach)                     | 1      | 0      | OK      | 0.06               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Callback called once (attach_us)                  | 1      | 0      | OK      | 0.07               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Callback not called when cancelled (attach)       | 1      | 0      | OK      | 0.09               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Callback not called when cancelled (attach_us)    | 1      | 0      | OK      | 0.08               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Callback override (attach)                        | 1      | 0      | OK      | 0.07               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Callback override (attach_us)                     | 1      | 0      | OK      | 0.07               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Multiple timeouts running in parallel (attach)    | 1      | 0      | OK      | 0.09               |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Multiple timeouts running in parallel (attach_us) | 0      | 0      | ERROR   | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Timing drift (attach)                             | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Timing drift (attach_us)                          | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Zero delay (attach)                               | 0      | 0      | SKIPPED | 0.0                |
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_drivers-lp_timeout | Zero delay (attach_us)                            | 0      | 0      | SKIPPED | 0.0                |
+-------------------------+-------------------+-------------------------------+---------------------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 14 SKIPPED / 7 OK / 1 ERROR
mbedgt: completed in 255.85 sec

```

CC : @maclobdell @bentcooke @ccli8

ccli8, 2018-09-10:
#8029 would fix the failure.

0xc0170, 2018-09-19:
5.10.1 will contain the fix, already on master

closed 2018-09-19 by 0xc0170