ARMmbed/mbed-os #8043: mbed_hal-sleep test fails on NUMAKER_PFM_M2351

studavekar, 2018-09-07:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_M2351

**Toolchain:**
ARMC6 

**mbed-os sha:**

6827c370 (HEAD -> master, origin/master, origin/feature-hal-spec-spi, origin/HEAD) Merge pull request #7971 from JammuKekkonen/fix_softdevice_memory_reservation_for_nrf52dk

**Steps to reproduce**

Use command run :  mbed test -m NUMAKER_PFM_M2351 -t ARMC6  -n tests-mbed_hal-sleep  -v

logs snippet
```
        calling mbedhtrun: mbedhtrun -m NUMAKER_PFM_M2351 -p /dev/ttyACM22:9600 -f "BUILD/tests/NUMAKER_PFM_M2351/ARMC6/TESTS/mbed_hal/sleep/sleep.hex" -e "TESTS/host_tests" -d /mnt/pci-0000_00_14_0-usb-0_7_2_3_1_0-scsi-0_0_0_0 -c default -t 13050000002D0101013B97B000000002 -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1536342497.37][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
[1536342497.39][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
[1536342497.39][HTST][INF] host test executor ver. 1.4.1
[1536342497.39][HTST][INF] copy image onto target...
[1536342497.39][COPY][INF] Waiting up to 60 sec for '13050000002D0101013B97B000000002' mount point (current is '/mnt/pci-0000_00_14_0-usb-0_7_2_3_1_0-scsi-0_0_0_0')...
[1536342497.67][mbedls.lstools_base]No valid file found to update JLink device details
[1536342497.67][mbedls.lstools_base]No valid file found to update JLink device details
[1536342497.67][mbedls.lstools_base]daplink entry: "0000" not found in platform database
[1536342510.11][HTST][INF] starting host test process...
[1536342510.11][CONN][INF] starting connection process...
[1536342510.11][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1536342510.11][CONN][INF] initializing serial port listener...
[1536342510.11][PLGN][INF] Waiting up to 60 sec for '13050000002D0101013B97B000000002' serial port (current is '/dev/ttyACM22')...
[1536342510.11][HTST][INF] setting timeout to: 60 sec
[1536342510.39][mbedls.lstools_base]No valid file found to update JLink device details
[1536342510.39][mbedls.lstools_base]No valid file found to update JLink device details
[1536342510.39][mbedls.lstools_base]daplink entry: "0000" not found in platform database
[1536342510.39][mbedls.lstools_base]No valid file found to update JLink device details
[1536342510.39][mbedls.lstools_base]daplink entry: "7012" not found in platform database
[1536342510.40][SERI][INF] serial(port=/dev/ttyACM22, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1536342510.40][SERI][INF] reset device using 'default' plugin...
[1536342510.51][SERI][INF] waiting 1.00 sec after reset
[1536342511.51][SERI][INF] wait for it...
[1536342511.51][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1536342511.51][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1536342511.51][CONN][INF] sending preamble '58f0f170-1284-4f61-af6d-92f6aa9716fc'
[1536342511.51][SERI][TXD] {{__sync;58f0f170-1284-4f61-af6d-92f6aa9716fc}}
[1536342511.64][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1536342511.70][CONN][INF] found SYNC in stream: {{__sync;58f0f170-1284-4f61-af6d-92f6aa9716fc}} it is #0 sent, queued...
[1536342511.70][HTST][INF] sync KV found, uuid=58f0f170-1284-4f61-af6d-92f6aa9716fc, timestamp=1536342511.694964
[1536342511.72][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1536342511.72][HTST][INF] DUT greentea-client version: 1.3.0
[1536342511.74][CONN][INF] found KV pair in stream: {{__timeout;60}}, queued...
[1536342511.74][HTST][INF] setting timeout to: 60 sec
[1536342511.78][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1536342511.78][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1536342511.78][HTST][INF] host test setup() call...
[1536342511.78][HTST][INF] CALLBACKs updated
[1536342511.78][HTST][INF] host test detected: default_auto

[1536342571.85][HTST][INF] test suite run finished after 60.11 sec...
[1536342571.86][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1536342571.90][HTST][INF] CONN exited with code: 0
[1536342571.90][HTST][INF] No events in queue
[1536342571.90][HTST][INF] stopped consuming events
[1536342571.90][HTST][INF] host test result(): None
[1536342571.90][HTST][WRN] missing __exit event from DUT
[1536342571.90][HTST][WRN] missing __exit_event_queue event from host test
[1536342571.90][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1536342571.90][HTST][INF] calling blocking teardown()
[1536342571.90][HTST][INF] teardown() finished
[1536342571.90][HTST][INF] {{result;timeout}}
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', u'NUMAKER_PFM_M2351', '-p', '/dev/ttyACM22:9600', '-f', u'"BUILD/tests/NUMAKER_PFM_M2351/ARMC6/TESTS/mbed_hal/sleep/sleep.hex"', '-e', u'"TESTS/host_tests"', '-d', u'/mnt/pci-0000_00_14_0-usb-0_7_2_3_1_0-scsi-0_0_0_0', '-c', 'default', '-t', u'13050000002D0101013B97B000000002', '-r', 'default', '-C', '4', '--sync', '5', '-P', '60'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-mbed_hal-sleep
        test case: tests-mbed_hal-sleep
mbedgt: test on hardware with target id: 13050000002D0101013B97B000000002
mbedgt: test suite 'tests-mbed_hal-sleep' ............................................................ TIMEOUT in 74.71 sec
        test case: 'tests-mbed_hal-sleep' ............................................................ ERROR in 74.71 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.5207175596
mbedgt: test suite report:
+-------------------------+-------------------+----------------------+---------+--------------------+-------------+
| target                  | platform_name     | test suite           | result  | elapsed_time (sec) | copy_method |
+-------------------------+-------------------+----------------------+---------+--------------------+-------------+
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_hal-sleep | TIMEOUT | 74.71              | default     |
+-------------------------+-------------------+----------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
+-------------------------+-------------------+----------------------+----------------------+--------+--------+--------+--------------------+
| target                  | platform_name     | test suite           | test case            | passed | failed | result | elapsed_time (sec) |
+-------------------------+-------------------+----------------------+----------------------+--------+--------+--------+--------------------+
| NUMAKER_PFM_M2351-ARMC6 | NUMAKER_PFM_M2351 | tests-mbed_hal-sleep | tests-mbed_hal-sleep | 0      | 1      | ERROR  | 74.71              |
+-------------------------+-------------------+----------------------+----------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 ERROR

```

CC : @maclobdell @bentcooke @ccli8

ccli8, 2018-09-10:
#8049 would fix the issue.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-425*

studavekar, 2018-10-12:
closing the ticket thanks @ccli8

closed 2018-10-12 by studavekar