ARMmbed/mbed-os #8045: STM32L053C8 UART Error in Mbed2 Latter Than Rev159

Tony-YI, 2018-09-09:
### Description

I am using a STM32L053C8 MCU and got some problems with it.

```
#include "mbed.h"
 
int main()
{
    DigitalOut  led(PA_14);
//    Serial      serial(PB_6, PB_7);

    while(1)
    {
        led = !led;
        wait(0.1);
    }
}
```

The code above works fine. However, when I un-comment the **Serial**, the LED **stops blinking**. As far as I know, both **(PB6, PB_7)** is the **UART1** serial port. I tried the **UART2** and **LPUART1** and both of them works fine. It looks like I can not define any **UART1** port.

According to my previous experience, I switched the **mbed-2** version to **Rev.159:7130f32** (I am using the online compiler) and un-comment the **Serial** and it **works fine**.

Jerome Coutant from ST said it is because the **UART1** is already used by the default STDIO. My question is that can I disable STDIO and use the **UART1** for my own purpose? Note that I am using the online compiler.

Thanks in advance.

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

Tony-YI, 2018-09-09:
After I un-commented the **Serial** line, I probed the **PB_4** pin (default LED1 on DISCO-L053C8 board) and its patterns matched the **mbed_die(void)** function in **platform/mbed_board.c**.

Still looking for a way to disable the STDIO or maybe make it using the UART2.

Tony-YI, 2018-09-09:
After some digging, it looks like it is caused by the following code in **/targets/TARGET_STM/serial_api.c**:

```
void serial_init(serial_t *obj, PinName tx, PinName rx)
{
    ...

    if ((tx == STDIO_UART_TX) || (rx == STDIO_UART_RX)) {
        stdio_config = 1;
    } else {
        if (uart_tx == pinmap_peripheral(STDIO_UART_TX, PinMap_UART_TX)) {
            error("Error: new serial object is using same UART as STDIO");
        }
    }
    ...
```

jeromecoutant, 2018-09-10:
Hi

Wiki page: https://os.mbed.com/teams/ST/wiki/STDIO

> So I need to redefine the STDIO_UART_TX and STDIO_UART_RX.

Yes, in a mbed_app.json file like :

"target_overrides": {     
    "DISCO_L053C8": {
         "target.stdio_uart_tx": "PB_6",
         "target.stdio_uart_rx": "PB_7"
     } }

Tony-YI, 2018-09-10:
Thanks @jeromecoutant , however, since I am using the online compiler, I suppose I do not have the mbed_app.json file.

jeromecoutant, 2018-09-10:
See for ex https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-wifi/

You can add a mbed_app.json file with online compiler

Tony-YI, 2018-09-10:
@jeromecoutant It works!!! Thanks a lot. You are really helpful!

jeromecoutant, 2018-09-10:
Why do you want to re-configure PB_6 and PB_7.
Thanks to the mbed_app.json, you new serial console is already configured.

Tony-YI, 2018-09-16:
> Why do you want to re-configure PB_6 and PB_7.
> Thanks to the mbed_app.json, you new serial console is already configured.

Dear @jeromecoutant, in my last comment, the line **Serial serial(PB_6, PB_7)** was commented. Then I probed the pin **PA_9** and **PB_6** respectively, and **PA_9** was outputting something instead of **PB_6** (mbed_app.json was added). This is contradictory to what we expect, i.e. **PB_6** is outputting something.

In fact, I have tried **(PA_14, PA_15)**,  **(PB_10, PB_11)** and **(PA_2, PA_3)** in **mbed_app.json**. I must add the following code so that the corresponding TX pin could output something:
```
FileHandle *mbed::mbed_override_console(int fd)
{
    static UARTSerial console(STDIO_UART_TX, STDIO_UART_RX, MBED_CONF_PLATFORM_STDIO_BAUD_RATE);
    return &console;
}
```
However, after I added the code above and tried **(PB_6, PB_7)**, the MCU went into **mbed_die()** routine. I am sure it was caused by the code below in  **/targets/TARGET_STM/serial_api.c**:
```
void serial_init(serial_t *obj, PinName tx, PinName rx)
{
    ...

    if ((tx == STDIO_UART_TX) || (rx == STDIO_UART_RX)) {
        stdio_config = 1;
    } else {
        if (uart_tx == pinmap_peripheral(STDIO_UART_TX, PinMap_UART_TX)) {
            error("Error: new serial object is using same UART as STDIO");
        }
    }
    ...
}
```
It looks like in **serial_api.c**, the **STDIO_UART_TX** and **STDIO_UART_RX** are still referring to **PA_9** and **PA_10** respectively.

Another interesting thing is that, I compile the above program using **mbed-cli (1.8.1)** and probed the **PA_9** and **PB_6**. None of them are outputting something.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6500*

jeromecoutant, 2019-06-17:
@Tony-YI Should we close this issue?

closed 2019-08-20 by 0xc0170

wubaojun1982, 2024-05-25:
My custom board is STM32F103ZCT6, there is the same problem. After defined like above, the problem seloved. Thx very much.

wubaojun1982, 2024-05-25:
My custom board is STM32F103ZCT6, there is the same problem. After methods mentioned like above, the problem solved. Thx very much.

wubaojun1982, 2024-05-25:
My custom board is STM32F103ZCT6, there is the same problem. After methods mentioned like above, the problem has been solved. Thx very much.