ARMmbed/mbed-os #8051: NRF52_DK build with bootloader doesn't run

JammuKekkonen, 2018-09-10:
### Description

App built to be run with bootloader on NRF52_DK doesn't start.

mbed_app.json:
`"NRF52_DK": {
            "target.mbed_app_start"             : "0xB400",
            "target.extra_labels_remove"        : ["SOFTDEVICE_S132_FULL", "SOFTDEVICE_COMMON"],
            "target.extra_labels_add"           : ["SOFTDEVICE_NONE"]
        }
`

- with mbed-os version 5.9.0 and 5.9.6 app doesn't run
- with commit #261f73fbf app runs
- with 5.8.6 says bootloader build not supported

Tested with GCC_ARM and ARM.

### Issue request type
[ ] Question  
[ ] Enhancement  
[x] Bug

JammuKekkonen, 2018-09-10:
@JuhaYlipekkala @yogpan01

JanneKiiskila, 2018-09-10:
@juhoeskeli  - time to look into this?

yogpan01, 2018-09-10:
@JammuKekkonen does it work with Mbed OS MASTER?

JanneKiiskila, 2018-09-10:
@ARMmbed/mbed-os-maintainers  - can get JIRA ticket linked in?