ARMmbed/mbed-os #8055: No network configuration for Network Socket test plan

MarceloSalazar, 2018-09-10:
Issue type: Bug 
SHA: master (920db63453b6aae7ccaa992fe7c9107f6664347f)

### Description
During 5.10 OOB, we're evaluating the Network Socket test plan document and steps to get started.
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md

I'm trying to compile tests for the K64F, following these steps:
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md#building-test-binaries

It seems this config in `mbed_app.json` is not enough
```
{
    "macros": ["MBED_EXTENDED_TESTS"]
}
```

I get an error during the compile step:
```
Compile [ 99.9%]: sleep.c
Compile [100.0%]: rtc_api.c
Building project wifi (K64F, GCC_ARM)
Scan: GCC_ARM
Scan: wifi
Compile [  5.3%]: get_interface.cpp
[Error] get_interface.cpp@21,2: #error [NOT_SUPPORTED] No network configuration found for this target.
[Error] get_interface.cpp@24,2: #error [NOT_SUPPORTED] Requires parameters from mbed_app.json
Building project emac (K64F, GCC_ARM)
```

Also, I've tried adding the default network interface but doesn't seem to work either.
```
{
    "macros": ["MBED_EXTENDED_TESTS"],
    "target_overrides": {
        "*": {
            "target.network-default-interface-type": "ETHERNET"
        }
    }
}
```

What might be missing?

@SeppoTakalo

SeppoTakalo, 2018-09-10:
Thank you for reporting this.

Apparently the test configuration is not working as I expected, or it has changed.
Do you know where the `mbed-os/tools/test-configs` would have been documented?

`mbed_app.json` apparently overrides that, but I cannot find any reference on how and when it happens.

SeppoTakalo, 2018-09-10:
Submitted a fix in https://github.com/ARMmbed/mbed-os/pull/8060

PR #8060 (2018-09-21): Clarify test configuration in Socket/Networking test document
### Description

Clarify test configuration in Socket/Networking test document.
Current documentation instructs user to supply `mbed_app.json` and rely content of `tools/test_configuration` to be appended. However, this is not working.
Full `mbed_app.json` needs to be provided.

Fixes #8055 

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-09-21 by 0xc0170