ARMmbed/mbed-os #8058: Unittest setup instructions should be expanded to include different environments

MarceloSalazar, 2018-09-10:
During 5.10 OOB, evaluated this doc.
https://github.com/ARMmbed/mbed-os/blob/master/UNITTESTS/README.md

Found there are missing/incomplete instructions to setup unittest environment.

- Windows
It should not be a strong requirement to use MinGW-W64 to run gcovr.
Many Windows users use this version of make (myself included).
http://gnuwin32.sourceforge.net/packages/make.htm

- brew on MacOS
Issues on setting up gcovr on default MacOS installation. There should be a known issues / troubleshooting section and pointers on how to solve common problems.
```
(pyenv):~/xxx$ pip install 'gcovr>=4.1'
Collecting gcovr>=4.1
  Could not fetch URL https://pypi.python.org/simple/gcovr/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590) - skipping
  Could not find a version that satisfies the requirement gcovr>=4.1 (from versions: )
No matching distribution found for gcovr>=4.1
```

- port on MacOS
Missing. Should be possible to let these users use gcovr.

[ ] Question  
[X] Enhancement  
[X] Bug

kegilbert, 2018-09-10:
@MarceloSalazar There's a troubleshooting section at the bottom of the page, do you think adding a link to the that section in the setup guide would help (provided the troubleshooting page is updated?)

lorjala, 2018-09-11:
@MarceloSalazar 

> It should not be a strong requirement to use MinGW-W64 to run gcovr. ...

MinGW and Gcovr target different use cases. MinGW is a port of GCC compilers for Windows and Gcovr is used to generate code coverage reports. 

Any port of GNU make and GCC compilers should work, but only MinGW-W64 is tested on Windows, because we cannot support all available tools and MinGW-W64 ships with both compilers and make.

> Issues on setting up gcovr on default MacOS installation.

What is your python and `pip --version`? We support pip version 10 or newer, same as Mbed CLI.

MarceloSalazar, 2018-09-11:
@kegilbert thanks for the PR. The changes and link to the troubleshooting is good :)

@lorjala this statement that has been added is great: "MinGW-W64 is recommended, although any windows port of the above GCC versions should work"

> What is your python and pip --version? We support pip version 10 or newer, same as Mbed CLI.

You're right, my environment is not correct. I'll work on this - let's ignore this for now.

0xc0170, 2018-09-19:
@MarceloSalazar Any outstanding work for this issue?

MarceloSalazar, 2018-09-19:
I think the main points have been addressed in PR #8069. So I'm happy to close this one.

However, there is a pending action (not for 5.10.0):

> Still looking into the missing port of gcovr on Mac.

@kegilbert do you think you could raise a ticket to keep track of this piece of work?

kegilbert, 2018-09-19:
Hi @MarceloSalazar 

Sorry for the delay! I wasn't able to reproduce the error, what version of python/pip were you running?

```
(pyenv) aus-daplink-mac-00:~ arm$ pip install 'gcovr>=4.1'
Collecting gcovr>=4.1
  Downloading https://files.pythonhosted.org/packages/65/3a/0b049c91cadaafa93cc7511aa2885b513e1ce71945a87d57858a06c9af47/gcovr-4.1-py2.py3-none-any.whl
Collecting jinja2 (from gcovr>=4.1)
  Using cached https://files.pythonhosted.org/packages/7f/ff/ae64bacdfc95f27a016a7bed8e8686763ba4d277a78ca76f32659220a731/Jinja2-2.10-py2.py3-none-any.whl
Collecting MarkupSafe>=0.23 (from jinja2->gcovr>=4.1)
Installing collected packages: MarkupSafe, jinja2, gcovr
Successfully installed MarkupSafe-1.0 gcovr-4.1 jinja2-2.10
```

MarceloSalazar, 2018-09-20:
@kegilbert my pip version is now 18.0 and I confirm everything works now on Mac!
I think it was caused by an old version of pip...

Closing this ticket - no further action is required. Thank a lot!

closed 2018-09-20 by MarceloSalazar