ARMmbed/mbed-os #8059: Running unittests: ERROR: Please specify target

MarceloSalazar, 2018-09-10:
Doing OOB 5.10 and evaluating unittest

Installed tools as per doc:
https://github.com/ARMmbed/mbed-os/blob/master/UNITTESTS/README.md

Cloned mbed-os (master)

`mbed test --unittests`

Error:

```
>mbed test --unittests
[mbed] WARNING: Could not find mbed program in current path "C:\mbed\OOB".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
[mbed] Auto-installing missing Python modules...
[mbed] ERROR: Please specify target using the -m switch or set default target using command 'mbed target'
---
```

ashok-rao, 2018-09-11:
@OPpuolitaival ..FYI..

lorjala, 2018-09-11:
@MarceloSalazar This is Mbed CLI specific issue, which should be fixed in 1.8.0. Is your mbed cli version 1.8.0?

OPpuolitaival, 2018-09-11:
https://github.com/ARMmbed/mbed-cli/commit/09462742d415fceaf62dd3631640d6558c72b193 fixed this one

SeppoTakalo, 2018-09-11:
Verified that this issue is not happening in mbed-cli 1.8.0

MarceloSalazar, 2018-09-11:
You're absolutely right, my bad!

closed 2018-09-11 by MarceloSalazar