ARMmbed/mbed-os #8068: Cannot initialize CAN on STM32F072RB

fjvva, 2018-09-10:
### Description

STM32F072RB Nucleo Board, using latest Mbed library.
To reproduce, past the following code in main.cpp, compile, and upload it. You will get an error output on USB Serial:

`#include "mbed.h"

//------------------------------------
// Hyperterminal configuration
// 9600 bauds, 8-bit data, no parity
//------------------------------------

Serial pc(SERIAL_TX, SERIAL_RX);
CAN canbus(PB_8, PB_9, 250000); 
 
int main() {
}`

### Issue request type

[ ] Question  
[ ] Enhancement  
[X ] Bug

jeromecoutant, 2018-09-11:
Hi @fjvva 
I am sorry, but I couldn't reproduce your issue...
Jerome

bcostm, 2018-09-11:
Hi,
Please precise:
    - What toolchain (name + version) are you using?
    - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
    - What is the SHA of Mbed OS (git log -n1 --oneline)?

fjvva, 2018-09-13:
I'm using the following:
-Mbed revision 169:a7c7l22 (Jun 2018)
-Mbed online compiler (web based)
-Code posted on first post

The MCU exact part number is STM32F072CBT6 in 48-pin package

bcostm, 2018-09-20:
I tried also to reproduce it today but it's OK with me too.

My conditions:
- NUCLEO_F072RB board
- mbed online compiler (running on Windows)
- mbed-os library revision 4466 (= 5.9.7) or mbed library revision 170 (= version 163 = 5.9.6)

I don't know what is wrong on your side ? Did you check again recently ?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6509*

bcostm, 2018-10-26:
@screamerbg OK to close this issue (not reproduced and no recent feedback received) ?  

ST_TO_BE_CLOSED

0xc0170, 2018-10-26:
@fjvva Please reopen with providing more information (if this issue still persists)

closed 2018-10-26 by 0xc0170

DOQnR, 2019-04-27:
You should terminate between CANH and CANL with 120 ohm. I think MBED's CAN class requires the closed CAN network when it initiates.

fjvva, 2019-04-27:
If you read my original message, the problem is not that the CAN message is not transmitted (which it is not), but that Mbed OS crashes while creating the CAN object. The termination resistor is used on the physical layer, and should only be placed in the termination of a bus (the end of the CAN wires). I have CAN working on LPC1768 with and without termination resistors, but the STM does not even bring it up.

fjvva, 2019-04-27:
The issue happens on a custom board that uses the ST part number STM32F072CBT6 part number, which is a QFN48 instead of a QFN64 package, though its the same product as the nucleo board i mention in the original post.

DOQnR, 2019-04-27:
I had the same problem. When I open between CANH and CANL MBED OS crashes while creating CAN object. But when I put 120 ohm resistor termination my custom board works. Although you have different problem, I recommend you to check your hardware.