ARMmbed/mbed-os #8080: Cellular/BC95: What is the best way to issue specific AT commands?

marcemmers, 2018-09-11:
### Description

We are working with a custom board with a Quectel BC68. The BC68 has the same commands as the BC95 which is supported. There are a couple of AT commands that are not supported by all module vendors but that I do need for this module, for example: AT+CCLK and AT+NCONFIG.

I am looking for the cleanest way to implement these AT commands.

It is no problem to implement them for the BC95 but then there is still no good way to get to that class because EasyCellularConnection only provides you with the base class CellularDevice. Or they need to be implemented in the base classes but then they need to be supported on all devices.

Another way could be to use the ATHandler class directly but I've ran into the same problem that it is not accessible. 

In short, what would be the best way to access the non-standard AT commands supported by the modules?

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

kivaisan, 2018-09-27:
@marcemmers Make an own cellular target which inherits QUECTEL_BC95. In your new target just override those methods where you need to use your specific AT commands.
Then just define CELLULAR_TARGET as your new cellular target.

Somethings like this:
```
class QUECTEL_BC68 : public QUECTEL_BC95 {
public:
    QUECTEL_BC68(events::EventQueue &queue);
    virtual ~QUECTEL_BC68();
protected: // AT_CellularDevice
    virtual AT_CellularPower *open_power_impl(ATHandler &at);
};
```
```
QUECTEL_BC68::QUECTEL_BC68(EventQueue &queue) : QUECTEL_BC95(queue) {}

QUECTEL_BC68::~QUECTEL_BC68() {}

AT_CellularPower *QUECTEL_BC68::open_power_impl(ATHandler &at)
{
    return new QUECTEL_BC68_CellularPower(at);
}
```
```
class QUECTEL_BC68_CellularPower : public QUECTEL_BC95_CellularPower {
public:
    QUECTEL_BC68_CellularPower(ATHandler &atHandler);
    virtual ~QUECTEL_BC68_CellularPower();

public: //from CellularPower
    virtual nsapi_error_t set_at_mode();
};
```
```
QUECTEL_BC68_CellularPower::QUECTEL_BC68_CellularPower(ATHandler &atHandler) : QUECTEL_BC95_CellularPower(atHandler) {}

QUECTEL_BC68_CellularPower::~QUECTEL_BC68_CellularPower() {}

nsapi_error_t QUECTEL_BC68_CellularPower::set_at_mode()
{
    tr_info("QUECTEL_BC68_CellularPower");
    return QUECTEL_BC95_CellularPower::set_at_mode();
}
```

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCELL-1374*

kivaisan, 2018-10-04:
As no activity in 7 days, I'll close this issue. @marcemmers if you still have any questions for this, feel free to reopen the issue.

closed 2018-11-21 by ciarmcom