ARMmbed/mbed-os #8084: NRF52_DK sleep/deep sleep profiling does not work

JuricaResetar, 2018-09-11:
### Description
Hi =)

[X ] Bug 
I'm trying to implement sleep/deep sleep profiling into my project on Nordic's NRF52_DK but with no success. I'm using official mbed-os version mbed-os-5.9.6 and mbed CLI with GCC_ARM compiler.
I've followed instructions from [mbed web](https://os.mbed.com/docs/v5.9/reference/power-management.html) and added MBED_SLEEP_TRACING_ENABLED macro into my mbed_app.json file but with no success.
I'm quite sure my printf and mbed_app.json part work because [this example](https://os.mbed.com/docs/v5.9/reference/mbed-statistics.html) for mbed statistics works great and the principle is the same (one macro and print on standard IO).

I have found the same problem description on mbed forum [here](https://os.mbed.com/questions/81928/Sleepdeep-sleep-profiling-tool-not-gener/).

I wunder if there's some kind of fix for that or should I have for new version of mbed-os?

Here's the main file (the one from official example on mbed-os):

```
#include "mbed.h"

int main()
{
	while(1)
	{
		// Deep sleep for 1 second
		printf("Deep sleep allowed: %i\r\n", sleep_manager_can_deep_sleep());
		wait(1.0);

		// Lock deep sleep
		printf("Locking deep sleep\r\n");
		sleep_manager_lock_deep_sleep();

		// Sleep for 1 second
		printf("Deep sleep allowed: %i\r\n", sleep_manager_can_deep_sleep());
		wait(1.0);
	}
}
```

TacoGrandeTX, 2018-09-12:
This is not working for two reasons:

1.  As mentioned in the forum post you will need to have `DEVICE_STDIO_MESSAGES` defined as this enables the print messages that the tracer produces.  
2. Even with `DEVICE_STDIO_MESSAGES` we noticed that the trace messages were stalled waiting on a mutex.  This is fixed by this PR which was just reviewed:  https://github.com/ARMmbed/mbed-os/pull/8048

Please bring in the above and try it again. 

We'll update the Forum post for those that aren't tracking GitHub.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1634*

closed 2018-10-23 by ciarmcom