ARMmbed/mbed-os #8087: Mutex failure while running tcpserver_accept test

aashishc1988, 2018-09-11:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
ARM

**Tests we are failing:**
TCPSERVER_ACCEPT  (Icetea)

**Steps to reproduce**

mbed test -m K64F -t ARM --icetea -n TCPSERVER_ACCEPT

**log snippet**
```
> 
cleaning up Icetea log directory.
10:59:24.460 Using Icetea version 1.0.1
10:59:24.460 Loading suite from file
10:59:24.460 Importing testcases for filtering
10:59:24.523 Filtering testcases
10:59:24.523 Suite creation complete.
10:59:24.539 Test cases prepared.
10:59:24.539 Starting suite.
10:59:24.539 Starting repeat 1 of 1
10:59:24.539 Iteration 1 of 1
10:59:24.539
10:59:24.539 START TEST CASE EXECUTION: 'TCPSERVER_ACCEPT'
10:59:24.539
10:59:24.539 | TC     MainThread: Start Test case 'TCPSERVER_ACCEPT'
10:59:24.539 | TC     MainThread: ====setUpTestBench====
10:59:24.555 | TC     MainThread: ====setUpTestBench====
10:59:24.561 | TC     MainThread: ---------------------
10:59:24.561 | TC     MainThread: Initialize DUT's connections
10:59:24.561 | RSP     MainThread: Allocating 2 duts.
10:59:24.607 | LAL     MainThread: Allocated device 0240000032044e45001e700a997b00136781000097969900
10:59:24.618 | LAL     MainThread: Allocated device 0240000032044e4500407009997b00446781000097969900
10:59:24.619 | RSP     MainThread: Allocation successful
10:59:24.632 | D1     Thread-5: Flashing dev: 0240000032044e45001e700a997b00136781000097969900
10:59:24.661 | FLS     Thread-5: sendBreak to device to reboot
10:59:24.924 | FLS     Thread-5: reset completed
        1 file(s) copied.
10:59:40.016 | FLS     Thread-5: sendBreak to device to reboot
10:59:40.276 | FLS     Thread-5: reset completed
10:59:40.717 | FLS     Thread-5: 0240000032044e45001e700a997b00136781000097969900 flash success
10:59:40.717 | TC     Thread-5: flash ready
10:59:40.717 | TC     Thread-5: Use board K64F as "D1" (id: 0240000032044e45001e700a997b00136781000097969900)
10:59:40.717 | D2     Thread-5: Flashing dev: 0240000032044e4500407009997b00446781000097969900
10:59:40.782 | FLS     Thread-5: sendBreak to device to reboot
10:59:41.059 | FLS     Thread-5: reset completed
        1 file(s) copied.
10:59:54.680 | FLS     Thread-5: sendBreak to device to reboot
10:59:54.943 | FLS     Thread-5: reset completed
10:59:55.397 | FLS     Thread-5: 0240000032044e4500407009997b00446781000097969900 flash success
10:59:55.397 | TC     Thread-5: flash ready
10:59:55.397 | TC     Thread-5: Use board K64F as "D2" (id: 0240000032044e4500407009997b00446781000097969900)
10:59:55.413 | D1     MainThread: DutSerial COM31

+----------+--------------+
| COM port |   Location   |
+----------+--------------+
|  COM31   | X = 0, Y = 0 |
+----------+--------------+
[... 126 lines cut ...]
11:03:31.865 | D2 --> DutThread: set --vt100 on
11:03:31.881 | D2 <-- DutThread: retcode: 0
11:03:31.900 | D2 --> DutThread: set --retcode false
11:03:31.951 | TC     MainThread: Test 'TCPSERVER_ACCEPT' FAIL, reason: COM31 CMD timeout: socket 0 accept 1
11:03:32.407 Test case TCPSERVER_ACCEPT failed, No retries left.

+------------------+---------+--------------------------------------+-------------+-----------+----------+
| Testcase         | Verdict |             Fail Reason              | Skip Reason | platforms | duration |
+------------------+---------+--------------------------------------+-------------+-----------+----------+
| TCPSERVER_ACCEPT |   fail  | COM31 CMD timeout: socket 0 accept 1 |             |    K64F   | 247.414  |
+------------------+---------+--------------------------------------+-------------+-----------+----------+
+---------------+----------------+
|    Summary    |                |
+---------------+----------------+
| Final Verdict |      FAIL      |
|     count     |       1        |
|    passrate   |     0.00 %     |
|      fail     |       1        |
|    Duration   | 0:04:07.414000 |
+---------------+----------------+
11:03:32.460 Cleanup done.
```
    [ ] Question
    [ ] Enhancement 
    [X ] Bug

aashishc1988, 2018-09-11:
@ashok-rao

ashok-rao, 2018-09-11:
@OPpuolitaival ..can you please check this? Thanks.

juhaylinen, 2018-09-12:
Fixed in https://github.com/ARMmbed/mbed-os/pull/8031

aashishc1988, 2018-09-12:
@juhaylinen I have verified that https://github.com/ARMmbed/mbed-os/pull/8031 fixes the issue and the test runs fine now. I will close the issue once the PR is merged.

closed 2018-09-17 by aashishc1988