ARMmbed/mbed-os #8088: Test was skipped but summary said PASS

aashishc1988, 2018-09-11:
While running a test suite using icetea, the test was skipped because required binary was not built, but at the end, the summary stated the final verdict as PASS. Shouldn't it be INCONCLUSIVE or something else that indicates a test was skipped.

Log:

```

cleaning up Icetea log directory.
11:48:55.851 Using Icetea version 1.0.1
11:48:55.851 Loading suite from file
11:48:55.855 Importing testcases for filtering
11:48:55.917 Filtering testcases
11:48:55.917 Suite creation complete.
11:48:55.933 Test cases prepared.
11:48:55.933 Starting suite.
11:48:55.933 Starting repeat 1 of 1
11:48:55.933 Iteration 1 of 1
11:48:55.933 TC 'test_cmdline' will be skipped because of 'Test requiring application binary not build'
11:48:56.404 Test case test_cmdline skipped.

+--------------+---------+-------------+---------------------------------------------+-----------+----------+
| Testcase     | Verdict | Fail Reason |                 Skip Reason                 | platforms | duration |
+--------------+---------+-------------+---------------------------------------------+-----------+----------+
| test_cmdline |   skip  |             | Test requiring application binary not build |           |    0     |
+--------------+---------+-------------+---------------------------------------------+-----------+----------+
+---------------+---------+
|    Summary    |         |
+---------------+---------+
| Final Verdict |   PASS  |
|     count     |    1    |
|    passrate   |  0.00 % |
|      skip     |    1    |
|    Duration   | 0:00:00 |
+---------------+---------+
11:48:56.466 Cleanup done.
```

[x ] Question  
[ ] Enhancement  
[ ] Bug

aashishc1988, 2018-09-11:
@OPpuolitaival @ashok-rao

jeromecoutant, 2018-09-12:
Hi
There is the same issue with greentea tests...
Macro TEST_SKIP_UNLESS_MESSAGE make the test "OK"

jonikula, 2018-09-12:
There is no overall verdict for the run to indicate that all tests were skipped. If it's something that's wanted we can add one.

aashishc1988, 2018-09-12:
@jupe So if I am running 30+ tests, considering that "final verdict" is the only parameter that tells me if test suite ran fine or not, lets say 3-4 tests skipped, if I am automating the part of gathering the test result information, verdict of PASS in that case is quite misleading. In my opinion, if any test was skipped, there is a valid reason behind it, so either we should remove that test as a part of suite, or give a verdict other than PASS. INCONCLUSIVE is a good option.

jupe, 2018-09-12:
I kind of agree with you. Like you say that if any test was skipped there is most probably good reason behind it. Currently `Icetea` have three ways to get skip test results:
* from test case meta-data which user has "hard-coded" to test
* from test suite
* if selected platform is not allowed for some particular test case (`allowed_platforms` -list in test meta-data)

In this case I think icetea integration to mbed-os is done so that generated test-suite mark test to be skipped if given binary is not build. Another option could be just omit that particular test from suite but then it would not be visible in results which tests was "skipped". I can also imagine that PASS final verdict in this case is just fine even there was some skipped tests - but if we want to change this behaviour @OPpuolitaival need to take control of this :)

jonikula, 2018-09-13:
So, what I'm reading here is that the action point from Icetea-point of view should be to change reporting to inconclusive if all tests were skipped. Further discussion needs to happen if we want to consider other changes as well.

OPpuolitaival, 2018-09-13:
This works as designed. This is very close to greentea behaviour. If we want want to change this, let's make a feature request and change greentea side too.

OPpuolitaival, 2018-09-13:
MBEDOSTEST-187

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-439*

closed 2018-11-26 by ciarmcom

jeromecoutant, 2018-12-13:
Hi

I fully agree with this issue description "Test was skipped but summary said PASS"
I understand that this issue is closed in the icetea context.

Do I need to open a new issue for Greentea ?

cmonr, 2018-12-13:
@jeromecoutant I think so. From what I gather, two more issues should be opened, one in Mbed OS, and another in Greentea, requesting the feature.

At least, that's what I gather from @OPpuolitaival comment:
> This is very close to greentea behaviour. If we want want to change this, let's make a feature request and change greentea side too.

aashishc1988, 2018-12-13:
I think this is where we are tracking this
MBEDOSTEST-200

jeromecoutant, 2018-12-13:
Add a +1 in your internal ticket for me! :-)