ARMmbed/mbed-os #8089: Error at running unittest: ip4string.h: No such file

MarceloSalazar, 2018-09-11:
### Description

I'm evaluating 5.10 during OOB and following instructions here
https://github.com/ARMmbed/mbed-os/blob/master/UNITTESTS/README.md

Mbed OS branch: mbed-os-5.10.0-rc1
Mbed CLI: 1.8.0
cmake version 3.12.2
gcc (GCC) 4.8.3
Windows 10

```
mbed test --unittests --coverage html

[mbed] Auto-installing missing Python modules...

#####################
Mbed OS unit testing:
#####################

Steps:

 [AUTO]         Prepare build directory
                 - Clean: False
                 - Directory: C:\mbed\BUILD\unittests

 [SET]          Build unit tests
                 - Debug: True
                 - CMake generator: Unix Makefiles
                 - Make program: gmake

 [SET]          Run unit tests
                 - Filter: None

 [SET]          Generate coverage reports
                 - Output: html
-- Configuring done

-- Generating done

-- Build files have been written to: C:/mbed/BUILD/unittests/googletest-download

[ 11%] Performing update step for 'googletest'

[ 22%] No configure step for 'googletest'

[ 33%] No build step for 'googletest'

[ 44%] No install step for 'googletest'

[ 55%] No test step for 'googletest'

[ 66%] Completed 'googletest'

[100%] Built target googletest

-- Configuring done

-- Generating done

-- Build files have been written to: C:/mbed/BUILD/unittests

[  1%] Building CXX object CMakeFiles/features_netsocket_TCPSocket.MbedOS.dir/C_/mbed/OOB/mbed-os/features/netsocket/SocketAddress.cpp.obj

C:/mbed/OOB/mbed-os/features/netsocket/SocketAddress.cpp:22:23: fatal error: ip4string.h: No such file or directory

 #include "ip4string.h"

                       ^

compilation terminated.

CMakeFiles/features_netsocket_TCPSocket.MbedOS.dir/build.make:63: recipe for target 'CMakeFiles/features_netsocket_TCPSocket.MbedOS.dir/C_/mbed/OOB/mbed-os/features/netsocket/SocketAddress.cpp.obj' failed

gmake[2]: *** [CMakeFiles/features_netsocket_TCPSocket.MbedOS.dir/C_/mbed/OOB/mbed-os/features/netsocket/SocketAddress.cpp.obj] Error 1

CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/features_netsocket_TCPSocket.MbedOS.dir/all' failed

gmake[1]: *** [CMakeFiles/features_netsocket_TCPSocket.MbedOS.dir/all] Error 2

Building unit tests failed.
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 2
       Path: "C:\mbed\OOB"
       Command: "c:\python27\python.exe C:\mbed\OOB\mbed-os\UNITTESTS\mbed_unittest.py --coverage html --build ..\BUILD\unittests"
       Tip: You could retry the last command with "-v" flag for verbose output

```
### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

birdhu01, 2018-09-11:
-- Generating done

-- Build files have been written to: .../oobtestingsuites/BUILD/unittests

Scanning dependencies of target gtest

[  1%] Building CXX object googletest-build/googlemock/gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o

[  3%] Linking CXX static library libgtest.a

[  3%] Built target gtest

[  4%] Building CXX object CMakeFiles/features_netsocket_UDPSocket.MbedOS.dir/.../oobtestingsuites/mbed-os/features/netsocket/SocketAddress.cpp.o

oobtestingsuites/mbed-os/features/netsocket/SocketAddress.cpp:22:10: fatal error: ip4string.h: No such file or directory

 #include "ip4string.h"

          ^~~~~~~~~~~~~

compilation terminated.

make[2]: *** [CMakeFiles/features_netsocket_UDPSocket.MbedOS.dir/...../oobtestingsuites/mbed-os/features/netsocket/SocketAddress.cpp.o] Error 1

make[1]: *** [CMakeFiles/features_netsocket_UDPSocket.MbedOS.dir/all] Error 2

make: *** [all] Error 2

Building unit tests failed.
       Code: 2
       Path: "/oobtestingsuites"
       Command: ".../oobtestingsuites/venv3.6/bin/python3 
../oobtestingsuites/mbed-os/UNITTESTS/mbed_unittest.py --build ./BUILD/unittests"

OPpuolitaival, 2018-09-12:
Fix: https://github.com/ARMmbed/mbed-os/pull/8086

0xc0170, 2018-09-19:
Will be in 5.10, fix integrated on master

closed 2018-09-19 by 0xc0170