ARMmbed/mbed-os #8098: Drivers don't tidy up HW afterwards

RobMeades, 2018-09-12:
### Description
I guess you're aware of this but several of the mbed drivers don't tidy up the HW when they have been deleted: `Serial()`, `I2C()` and `Timer()` (see #7847) are the examples I know of.   There are hints (e.g. in [serial_api.h](https://github.com/ARMmbed/mbed-os/blob/master/hal/serial_api.h#L121) that this is intended but it is not implemented or, in the `I2C()` case, not even hinted at.

This means that those HW blocks continue to run and suck power, which is not ideal for low power applications.

Is there a plan/schedule to resolve this in a future mbed-os release?

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

deepikabhavnani, 2018-09-13:
@RobMeades - Thanks for reporting the issue, mbed-os HAL team is aware of this, and working on this  with each HAL+driver API updates. 
Since deletion/cleanup of HW needs to be implemented for each and every target it will be bit slow. 

Related to https://github.com/ARMmbed/mbed-os/issues/3106.

deepikabhavnani, 2018-09-13:
Closing this. Follow https://github.com/ARMmbed/mbed-os/issues/3106 for tracking.

closed 2018-09-13 by deepikabhavnani

RobMeades, 2018-09-14:
Understood, thanks.