ARMmbed/mbed-os #8101: MBED-OS 5.10 OOB - export to make_armc5, error in linking phase

zvikas, 2018-09-12:
I am using https://github.com/ARMmbed/mbed-os-example-filesystem example and testing export on Windows OS (windows cmdline).
My export command: mbed export -i make_armc5 -m K64F
The export finished and I use "make" command for building (with make clean and without) and got the following error in linking (compilation finished):
 
make (e=87): The parameter is incorrect.
C:/work/oob5.10/mbed-os-example-filesystem/Makefile:1229: recipe for target 'mbed-os-example-filesystem.elf' failed
make[1]: *** [mbed-os-example-filesystem.elf] Error 87
Makefile:25: recipe for target 'all' failed
make: *** [all] Error 2

complete log:
[error.txt](https://github.com/ARMmbed/mbed-os/files/2375568/error.txt)

zvikas, 2018-09-12:
the same issue occurs when exporting to make_iar

zvikas, 2018-09-12:
Tested on both 5.10 OOB rc1 and rc2 branch/tags

theotherjimmy, 2018-09-13:
@zvikas You got it: it's command line length related. There is no workaround at this time, and It's a documented issue.

JojoS62, 2018-09-13:
haven't you tried to do a workaround with response files for the linker?
https://gcc.gnu.org/wiki/Response_Files
This problem affects also the IDE exports, but I'm afraid its more dificult to solve there because the makefile is generated by the IDE.

0xc0170, 2018-09-19:
Duplicate https://github.com/ARMmbed/mbed-os/issues/6335 , please refer to this issue. We will close this one.

closed 2018-09-19 by 0xc0170