ARMmbed/mbed-os #8105: [OoB_5.10]: mbed dm init fails with invalid API key.

ashok-rao, 2018-09-12:
### Description

I have an Admin API key from the staging Cloud portal : OS2.

While OoB'ing mbed dm (RC2), I run the following command:
```mbed device-management init -d arm.com --model-name ashok --force -q -v```

But it fails with:
```
C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal>mbed device-management init -d arm.com --model-name ashok --force -q -v
[mbed] Working path "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal" (program)
[mbed] Auto-installing missing Python modules...
[mbed] Exec "c:\python27\python.exe -u C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py init -d arm.com --model-name ashok --force -q" in "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal"
Traceback (most recent call last):
  File "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py", line 155, in <module>
    main()
  File "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py", line 150, in main
    }[options.action](options) or 0
  File "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py", line 86, in inner
    'key': getenv("MBED_CLOUD_SDK_API_KEY")
  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 150, in next
    return self.__next__()
  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 155, in __next__
    self._get_next_page()
  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 96, in _get_next_page
    resp = self._func(**query)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\apis\developer_api.py", line 446, in get_all_api_keys
    (data) = self.get_all_api_keys_with_http_info(**kwargs)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\apis\developer_api.py", line 531, in get_all_api_keys_with_http_info
    collection_formats=collection_formats)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 353, in call_api
    _return_http_data_only, collection_formats, _preload_content, _request_timeout)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 113, in wrapped_f
    return fn(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 182, in __call_api
    _request_timeout=_request_timeout)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 374, in request
    headers=headers)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\rest.py", line 235, in GET
    query_params=query_params)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\rest.py", line 226, in request
    raise ApiException(http_resp=r)
mbed_cloud._backends.iam.rest.ApiException: (401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Length': '145', 'Access-Control-Expose-Headers': 'X-Request-ID,Content-Length,Content-Disposition', 'Set-Cookie': 'TS017f188d=01c6170234cf57758fd9698227189a67587495b9711905a002bee0a298ef658c294905f42c243ab2c85155aaf7f3ad4926c4a7b2be; Path=/', 'Strict-Transport-Security': 'max-age=10886400; includeSubDomains; preload', 'Vary': 'Origin', 'X-Request-ID': '00005b993c794eb641c93ce200000000', 'Connection': 'keep-alive', 'Date': 'Wed, 12 Sep 2018 16:19:06 GMT', 'Access-Control-Allow-Origin': 'https://portal.mbedcloud.com', 'Content-Type': 'application/json; charset=utf-8'})
HTTP response body: {"request_id":"00005b993c794eb641c93ce200000000","object":"error","message":"The provided API key is invalid","code":401,"type":"invalid_apikey"}

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal"
       Command: "c:\python27\python.exe -u C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py init -d arm.com --model-name ashok --force -q"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

because: https://portal.mbedcloud.com is hardcoded somewhere. I & our customers who have on-prem deployments or a different URL to their device management server will need to configure this - either via CLI or will atleast need docs as to where this can be configured.

We need to make this as a configurable param or need to add docs for where to find it.

cc: @theotherjimmy @teetak01 

### Issue request type
[ ] Question  
[ ] Enhancement  
[x] Bug

teetak01, 2018-09-13:
@ashok-rao fix already here https://github.com/ARMmbed/mbed-os/pull/8063

theotherjimmy, 2018-09-14:
@ashok-rao It's not hard coded. Please use `mbed config -G CLOUD_HOST` to configure it.

ashok-rao, 2018-09-17:
@theotherjimmy & @teetak01 : Unfortunately doesn't seem to fix .. here's logs:
```
C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal>mbed config -G --list
[mbed] Global config:
MBED_CLOUD_SDK_HOST=https://portal-os2.mbedcloudstaging.net
CACHE=on
CLOUD_HOST=https://api-os2.mbedcloudstaging.net
CLOUD_SDK_API_KEY=ak_1MDE1YjVjO <snip>

[mbed] Local config (C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal):
TOOLCHAIN=GCC_ARM
TARGET=K64F
```

And the error to this cmd: ``` mbed dm init -d "arm.com" --model-name "ashok" --force -q ```

```
C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal>mbed config -G CLOUD_HOST https://api-os2.mbedcloudstaging.net
[mbed] https://api-os2.mbedcloudstaging.net now set as global CLOUD_HOST

C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal>mbed dm init -d "arm.com" --model-name "ashok" --force -q
Traceback (most recent call last):
  File "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py", line 167, in <module>
    main()
  File "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py", line 162, in main
    }[options.action](options) or 0
  File "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py", line 99, in inner
    'key': api_key
  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 150, in next
    return self.__next__()
  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 155, in __next__
    self._get_next_page()
  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 96, in _get_next_page
    resp = self._func(**query)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\apis\developer_api.py", line 446, in get_all_api_keys
    (data) = self.get_all_api_keys_with_http_info(**kwargs)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\apis\developer_api.py", line 531, in get_all_api_keys_with_http_info
    collection_formats=collection_formats)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 353, in call_api
    _return_http_data_only, collection_formats, _preload_content, _request_timeout)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 113, in wrapped_f
    return fn(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 182, in __call_api
    _request_timeout=_request_timeout)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 374, in request
    headers=headers)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\rest.py", line 235, in GET
    query_params=query_params)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\rest.py", line 226, in request
    raise ApiException(http_resp=r)
mbed_cloud._backends.iam.rest.ApiException: (401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Length': '145', 'Access-Control-Expose-Headers': 'X-Request-ID,Content-Length,Content-Disposition', 'Set-Cookie': 'TS017f188d=01d57244a42e28c7eb1b5a4002aee0614bc99c985d3a3ec8a570d3941ad06395bfb7aae93289817fd24a84d91c96d3c86b37c8baec; Path=/', 'Strict-Transport-Security': 'max-age=10886400; includeSubDomains; preload', 'Vary': 'Origin', 'X-Request-ID': '00005b9f959bf2a5746915c200000007', 'Connection': 'keep-alive', 'Date': 'Mon, 17 Sep 2018 11:53:00 GMT', 'Access-Control-Allow-Origin': 'https://portal.mbedcloud.com', 'Content-Type': 'application/json; charset=utf-8'})
HTTP response body: {"request_id":"00005b9f959bf2a5746915c200000007","object":"error","message":"The provided API key is invalid","code":401,"type":"invalid_apikey"}

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal"
       Command: "c:\python27\python.exe -u C:\Ashok\mbed_CLI\OoB_5_10\PDM\mbed-cloud-client-example-sources-internal\mbed-os\tools\device_management.py init -d arm.com --model-name ashok --force -q"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-456*

ashok-rao, 2019-01-29:
Not relevant anymore. Closing.

closed 2019-01-29 by ashok-rao