ARMmbed/mbed-os #8107: UNITTEST Warnings: Windows path separators, variable 'tchain'

yennster, 2018-09-12:
### Description

Environment: Windows 10
gcc mingw version: 8.1.0
Mbed OS version: mbed-os-5.10.0-rc2
Mbed CLI version: 1.8.0

After running command: `mbed test --unittests --coverage html`

Warnings:
```
100% tests passed, 0 tests failed out of 7

Total Test time (real) =   4.17 sec

(WARNING) filters must use forward slashes as path separators

(WARNING) your filter : C:\Users\jenplu01\Repos\oob-5.10\mbed-os\UNITTESTS

(WARNING) did you mean: C:\Users/jenplu01/Repos/oob-5.10/mbed-os\UNITTESTS

(WARNING) filters must use forward slashes as path separators

(WARNING) your filter : C:\Users\jenplu01\Repos\oob-5.10\mbed-os\BUILD\unittests

(WARNING) did you mean: C:\Users/jenplu01/Repos/oob-5.10/mbed-os\BUILD\unittests

HTML code coverage report created.
[mbed] ERROR: Unknown Error: local variable 'tchain' referenced before assignment
---
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

MarceloSalazar, 2018-09-13:
@OPpuolitaival please review

0xc0170, 2018-09-19:
@OPpuolitaival This should be fixed (two fixes above were merged) and are part of 5.10.

closed 2018-09-19 by 0xc0170