ARMmbed/mbed-os #8110: Possible error of LPUART RCC setting in serial_baud

Eman869, 2018-09-13:
### Description

The RCC configuration looks wrong while setting serial_buad in STM devices LPUART.

In mbed-os/targets/TARGET_STM/serial_api.c

#if ((MBED_CONF_TARGET_LPUART_CLOCK_SOURCE) & USE_LPUART_CLK_LSE)
        if (baudrate <= 9600) {
            // Enable LSE in case it is not already done
            if (!__HAL_RCC_GET_FLAG(RCC_FLAG_LSERDY)) {
                RCC_OscInitTypeDef RCC_OscInitStruct = {0};
                RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSE;
                RCC_OscInitStruct.HSIState       = RCC_LSE_ON;
                RCC_OscInitStruct.PLL.PLLState   = RCC_PLL_OFF;
                HAL_RCC_OscConfig(&RCC_OscInitStruct);
            }

RCC_OscInitStruct.HSIState       = RCC_LSE_ON;
should be 
RCC_OscInitStruct.LSEState       = RCC_LSE_ON;

Am I right?
 
### Issue request type

[ ] Question  
[ ] Enhancement  
[x ] Bug

0xc0170, 2018-09-19:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-09-19:
Humm I think you're right. A bad copy/paste from HSI code below...

Thanks to report this bug. We'll fix it ASAP.

PR #8187 (2018-09-21): STM32: fix wrong LSE config in serial_baud function
### Description

The LSE configuration was incorrectly done in serial_baud function.

Fixes #8110

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-09-21 by 0xc0170