ARMmbed/mbed-os #8111: Cannot wake up STM32L476 by LPUART from sleep / deepsleep.

Eman869, 2018-09-13:
### Description

There are problems to wake up STM32L476 by LPUART. I made a small project and would like to wake up STM32L476 by LPUART RXNE interrupt. The STM32L476RG Nucleo board would connect to another board thru LPUART PC0/PC1. I expected to see system wake up after any word sent from the companion board. However, system does not wake up. I had noticed that the power consumption is not correct during sleep / deepsleep. It costs 200uA. It does not match power consumption when STM32L476 is in sleep or STOP 2 mode. The project for reference is attached.

[WakeUpByLPUART.zip](https://github.com/ARMmbed/mbed-os/files/2378168/WakeUpByLPUART.zip)

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

bcostm, 2018-09-18:
> system does not wake up

There is no automatic tests in place to check this. So, there should be something missing somewhere that prevent this mode to work correctly. We'll have to work on this use case.

> I had noticed that the power consumption is not correct during sleep / deepsleep. It costs 200uA. It does not match power consumption when STM32L476 is in sleep or STOP 2 mode.

Maybe due to leakage in some GPIOs ? Check that all unused GPIOs and Debug IOs are configured in Analog Input. You can use ST HAL functions for this.

Eman869, 2018-09-18:
@bcostm I had compared with choosing LSE as LPUART clock source, with the same project, and the power consumption is 8uA. The difference of power consumption is quite huge when choosing different clock source of LPUART, even with the same GPIO mode and debug IO attached.

screamerbg, 2018-09-20:
CC @jeromecoutant

Eman869, 2018-09-21:
@bcostm The power consumption issue is caused by HSI clock is enabled in STOP mode. It causes extra power consumption. Only LSE clock should be enabled in STOP mode. I modified the source code in below:

In mbed-os/targets/TARGET_STM/serial_api.c

#if defined(LPUART1_BASE)
    if (huart->Instance == LPUART1) {
        if (obj_s->baudrate <= 9600) {
#ifdef (USE_LPUART_CLK_LSE)			
            HAL_UARTEx_EnableClockStopMode(huart);
#endif
            HAL_UARTEx_EnableStopMode(huart);
        } else {
            HAL_UARTEx_DisableClockStopMode(huart);
            HAL_UARTEx_DisableStopMode(huart);
        }
    }
#endif

    Power consumption is back to 8uA in STOP mode. However, system still cannot wake up from STOP mode.

bcostm, 2018-09-27:
I have tested this basic example on the NUCLEO_L476RG:

```
#include "mbed.h"

Serial pc(PC_1, PC_0); // Connect CN3.TX to PC_0 (A0) and CN3.RX to PC_1 (A1)

DigitalOut led(LED1);

void callback() {
    // Note: you need to actually read from the serial to clear the RX interrupt
    printf("(%c)\n", pc.getc()); // Use SERIAL_TX/RX pins
    led = !led;
}

int main()
{
    pc.attach(&callback);
    printf("Press any character to toggle the led\n"); // Use SERIAL_TX/RX pins
    while(1) {
    }
}
```

mbed online compiler + mbed lib rev 163 (= mbed-os-5.9.6) : OK
mbed CLI + mbed-os-5.9.7 : OK
mbed CLI + mbed-os-5.10.0 : **FAIL**

I've got this error:
++ MbedOS Error Info ++
Error Status: 0x80020115 Code: 277 Module: 2
Error Message: Mutex lock failed
Location: 0x8006091
Error Value: 0xFFFFFFFA
Current Thread: Id: 0x20001F20 Entry: 0x800755D StackSize: 0x1000 StackMem: 0x20000F20 SP: 0x20017F70
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020115

I have no idea what has been broken in the interrupts ?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6551*

jeromecoutant, 2019-06-17:
@0xc0170 I think we can close this issue

closed 2019-06-27 by MarceloSalazar