ARMmbed/mbed-os #8113: error: Microsoft Visual C++ 14.0 is required

MarceloSalazar, 2018-09-13:
### Description
Evaluating 5.10 during OOB round 2.
Found error "Microsoft Visual C++ 14.0 is required" while trying to get started with Mbed OS.

Environment:
- Mbed CLI 1.8.0
- Python 3.7.0
- Windows 10

Installing Mbed OS in new PC and hitting errors:

```
mbed import test
cd test\mbed-os
mbed update mbed-os-5.10.0-rc2

[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Updating library "mbed-os" to branch/tag "mbed-os-5.10.0-rc2"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#dd2d15979665a3d70b18cf722ca91caf4e4ee523"
[mbed] Auto-installing missing Python modules...
[mbed] WARNING: Unable to auto-install required Python modules.
       The mbed OS tools in this program require the following Python modules: git, mbed_cloud_sdk, icetea
       You can install all missing modules by running "pip install -r requirements.txt" in "C:\mbed\OOB2\test2\mbed-os"
---

C:\mbed\OOB2\test2\mbed-os>pip install -r requirements.txt
Collecting git+https://github.com/armmbed/manifest-tool.git@v1.4.5 (from -r requirements.txt (line 18))
  Cloning https://github.com/armmbed/manifest-tool.git (to revision v1.4.5) to c:\users\marsal01\appdata\local\temp\pip-req-build-rx5x5sau
Requirement already satisfied: colorama>=0.3.3 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 1)) (0.3.9)
Requirement already satisfied: PySerial>=2.7 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 2)) (3.4)
Requirement already satisfied: PrettyTable>=0.7.2 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 3)) (0.7.2)
Requirement already satisfied: Jinja2>=2.7.3 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 4)) (2.10)
Requirement already satisfied: IntelHex>=1.3 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 5)) (2.2.1)
Requirement already satisfied: junit-xml in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 6)) (1.8)
Requirement already satisfied: pyYAML in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 7)) (3.13)
Requirement already satisfied: requests in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 8)) (2.19.1)
Requirement already satisfied: mbed-ls==1.*,>=1.4.2 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 9)) (1.6.0)
Requirement already satisfied: mbed-host-tests>=1.1.2 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 10)) (1.4.1)
Requirement already satisfied: mbed-greentea>=0.2.24 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 11)) (1.4.0)
Requirement already satisfied: beautifulsoup4>=4 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 12)) (4.6.3)
Requirement already satisfied: fuzzywuzzy>=0.11 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 13)) (0.17.0)
Requirement already satisfied: pyelftools>=0.24 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 14)) (0.25)
Requirement already satisfied: jsonschema>=2.6 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 15)) (2.6.0)
Requirement already satisfied: future>=0.16.0 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 16)) (0.16.0)
Requirement already satisfied: six>=1.11.0 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 17)) (1.11.0)
Requirement already satisfied: mbed-cloud-sdk==2.0.0 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 19)) (2.0.0)
Collecting icetea<2,>=1.0.1 (from -r requirements.txt (line 20))
  Using cached https://files.pythonhosted.org/packages/66/b9/74d07788ebb2c1a0988479cdb5615ae6d122899121c0d54f210a360992f1/icetea-1.0.1.tar.gz
Collecting ecdsa>=0.13 (from manifest-tool==1.4.5->-r requirements.txt (line 18))
  Using cached https://files.pythonhosted.org/packages/63/f4/73669d51825516ce8c43b816c0a6b64cd6eb71d08b99820c00792cb42222/ecdsa-0.13-py2.py3-none-any.whl
Collecting cryptography>=2.0.0 (from manifest-tool==1.4.5->-r requirements.txt (line 18))
  Using cached https://files.pythonhosted.org/packages/da/f6/6b9f031611697cb61d25c884237a8762725df7373a0accd49b39fbbb23cd/cryptography-2.3.1-cp37-cp37m-win32.whl
Collecting pyasn1<0.3.0,>=0.2.1 (from manifest-tool==1.4.5->-r requirements.txt (line 18))
  Using cached https://files.pythonhosted.org/packages/a5/ae/6b4c4cb9420edddd7401782f55504130d1269f2e5ae3ba3c986da167dd6c/pyasn1-0.2.3-py2.py3-none-any.whl
Collecting asn1ate>=0.5 (from manifest-tool==1.4.5->-r requirements.txt (line 18))
  Using cached https://files.pythonhosted.org/packages/ec/20/7310475955a893ddd0beb7c8a9b1f940390d67bc46cae9d60c7bc7566d5c/asn1ate-0.6.0.tar.gz
Collecting pyparsing>=2.1.0 (from manifest-tool==1.4.5->-r requirements.txt (line 18))
  Using cached https://files.pythonhosted.org/packages/6a/8a/718fd7d3458f9fab8e67186b00abdd345b639976bc7fb3ae722e1b026a50/pyparsing-2.2.0-py2.py3-none-any.whl
Requirement already satisfied: urllib3>=1.20 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from manifest-tool==1.4.5->-r requirements.txt (line 18)) (1.23)
Collecting protobuf<3.6.0,>=3.5.0 (from manifest-tool==1.4.5->-r requirements.txt (line 18))
  Using cached https://files.pythonhosted.org/packages/bf/dc/42d35afffe88ce301c90457eb60d145dd64144f283ef4bf927c1187ef114/protobuf-3.5.2.post1-py2.py3-none-any.whl
Requirement already satisfied: MarkupSafe>=0.23 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from Jinja2>=2.7.3->-r requirements.txt (line 4)) (1.0)
Requirement already satisfied: idna<2.8,>=2.5 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from requests->-r requirements.txt (line 8)) (2.7)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from requests->-r requirements.txt (line 8)) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from requests->-r requirements.txt (line 8)) (2018.8.24)
Requirement already satisfied: fasteners in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from mbed-ls==1.*,>=1.4.2->-r requirements.txt (line 9)) (0.14.1)
Requirement already satisfied: appdirs>=1.4 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from mbed-ls==1.*,>=1.4.2->-r requirements.txt (line 9)) (1.4.3)
Requirement already satisfied: pyOCD>=0.8.1a1 in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from mbed-host-tests>=1.1.2->-r requirements.txt (line 10)) (0.12.0)
Requirement already satisfied: lockfile in c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages (from mbed-greentea>=0.2.24->-r requirements.txt (line 11)) (0.12.2)
[... 57 lines cut ...]
    copying trollius\streams.py -> build\lib.win32-3.7\trollius
    copying trollius\subprocess.py -> build\lib.win32-3.7\trollius
    copying trollius\tasks.py -> build\lib.win32-3.7\trollius
    copying trollius\test_support.py -> build\lib.win32-3.7\trollius
    copying trollius\test_utils.py -> build\lib.win32-3.7\trollius
    copying trollius\time_monotonic.py -> build\lib.win32-3.7\trollius
    copying trollius\transports.py -> build\lib.win32-3.7\trollius
    copying trollius\unix_events.py -> build\lib.win32-3.7\trollius
    copying trollius\windows_events.py -> build\lib.win32-3.7\trollius
    copying trollius\windows_utils.py -> build\lib.win32-3.7\trollius
[... cut ...] p.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\marsal01\AppData\Local\Temp\pip-record-ieoldviq\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\marsal01\AppData\Local\Temp\pip-install-flcn6c3f\trollius\

```

Summary, I should not need to install 3rd party tools, such Microsoft Visual C++.

[ ] Question  
[ ] Enhancement  
[X] Bug

MarceloSalazar, 2018-09-13:
Workaround: Install "Microsoft Visual C++ Build Tools"

- The link shown in the console (http://landinghub.visualstudio.com/visual-cpp-build-tools) doesn't work. Go to this page instead: https://visualstudio.microsoft.com/visual-cpp-build-tools/
- Install "Build Tools for Visual Studio 2017" as indicated.
- In Visual Studio installer, Workloads, select "Visual C++ build tools" (~3.6GB)
- Reboot

theotherjimmy, 2018-09-13:
@OPpuolitaival Looks like you have another native-install-using dep. I thought we got rid of pyshark.

OPpuolitaival, 2018-09-13:
@theotherjimmy you are right. There is pyshark library in icetea dependencies which depends on trollius that need compiling. We need to make whole wireshark thing optional

OPpuolitaival, 2018-09-13:
@jonikula can you make wireshark optional?

jonikula, 2018-09-14:
I guess we have to.

jonikula, 2018-09-14:
I just pushed this PR. https://github.com/ARMmbed/icetea/pull/43

It very roughly pulls out the entire wireshark support. If some tests need that, then this will break them. If we want to support those tests, more time to support the feature is needed.

jupe, 2018-09-14:
Here is bit better approach: https://github.com/ARMmbed/icetea/pull/44 (does not drop feature - just change optionally it).

jupe, 2018-09-14:
There was actually bug in icetea that tries to install `pyshark-legacy`for python 3.x which are not even compatible - which causing this installation failure mentioned in description in first place.  `pyshark` -module instead are compatible with python 3.x and should not need any compilers. In case of python2.x pyshark-legacy should contains all pre-compiled packages so it doesn't need compilers either unless some exotic environment is in use. So I'm pretty sure that https://github.com/ARMmbed/icetea/pull/34 fixes this issue and no needs for ARMmbed/icetea#44  at all.

jupe, 2018-09-14:
@MarceloSalazar, could you verify that icetea#c16fec2c34f5362306fd4b3717a1a1c9306ccf44 does not require anymore visual-studio when installing ?

MarceloSalazar, 2018-09-17:
@jupe confirmed c16fec2c34f5362306fd4b3717a1a1c9306ccf44 doesn't seem to require visual-studio when tested in a fresh Win10/Python 3.6 PC. Thanks!

closed 2018-09-17 by cmonr