ARMmbed/mbed-os #8115: Hard fault with mesh applications when using MCR20A RF with IDE

artokin, 2018-09-13:
### Description

Hard fault detected when testing https://github.com/ARMmbed/mbed-os-example-mesh-minimal or https://github.com/ARMmbed/nanostack-border-router with mbed-os-5.10.0-rc2 branch and MCR20A RF.

Steps to reproduce with nanostack-border-router:
- checkout mbed-os-5.10.0-rc2 branch from the application
- Make `mbed deploy`
- Modify file `./configs/Thread_Atmel_RF.json` to use MCR20A RF
- Export build ` mbed export -i uvision5 -m k64f --app-config ./configs/Thread_Atmel_RF.json`
- Build with Keil uvision
- Run the application using Keil
- Observe following outcome from the serial window:

```
++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x2000155C, Not allowed in ISR context
Location: 0x730B
Error Value: 0x2000155C
Current Thread: Id: 0x20003440 Entry: 0x2F729 StackSize: 0x1000 StackMem: 0x20002440 SP: 0x2002FEB8
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80010133
-- MbedOS Error Info --
```

The issue can be reproduced in the same way in mbed-os-example-mesh-minimal:
- `mbed export -i IAR -m k64f --app-config ./configs/mesh_thread_MCR20A.json`
- compile using IAR Workbench
- Observe the result:

```
++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20003754, Not allowed in ISR context
Location: 0x0
Error Value: 0x20003754
Current Thread: Id: 0x200000F0 Entry: 0x21AC5 StackSize: 0x1000 StackMem: 0x20000150 SP: 0x2002FEC8
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80010133
-- MbedOS Error Info --
```
[image]

The error can be reproduced only by exporting and IDE compilation. Compilations done in command line are working fine.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

TuomoHautamaki, 2018-10-03:
Fix in PR https://github.com/ARMmbed/mbed-os/pull/8243

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTTHD-2818*

TuomoHautamaki, 2018-10-11:
PR now merged

closed 2018-10-23 by ciarmcom