ARMmbed/mbed-os #8116: Traceback error when using Icetea with Python 3

MarceloSalazar, 2018-09-13:
### Description

Evaluating 5.10 during OOB round 2.
Found Traceback error

Environment:

- Mbed CLI 1.8.0
- Python 3.7.0
- Both: Windows 10 and OSX El Capitan

```
>icetea --version
Traceback (most recent call last):
  File "C:\Users\marsal01\AppData\Local\Programs\Python\Python37-32\Scripts\icetea-script.py", line 11, in <module>
    load_entry_point('icetea==1.0.1', 'console_scripts', 'icetea')()
  File "c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages\icetea_lib\main.py", line 20, in icetea_main
    from icetea_lib import IceteaManager
  File "c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages\icetea_lib\IceteaManager.py", line 35, in <module>
    from icetea_lib.TestSuite.TestSuite import TestSuite, SuiteException
  File "c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages\icetea_lib\TestSuite\TestSuite.py", line 33, in <module>
    from icetea_lib.TestSuite.TestcaseContainer import TestStatus, DummyContainer
  File "c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages\icetea_lib\TestSuite\TestcaseContainer.py", line 29, in <module>
    from icetea_lib.bench import ReturnCodes
  File "c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages\icetea_lib\bench.py", line 53, in <module>
    from icetea_lib.DeviceConnectors.Dut import DutConnectionError
  File "c:\users\marsal01\appdata\local\programs\python\python37-32\lib\site-packages\icetea_lib\DeviceConnectors\Dut.py", line 330
    self.execute_command(cli_cmd[0], wait=wait, async=asynchronous)
                                                    ^
SyntaxError: invalid syntax
```

[ ] Question  
[ ] Enhancement  
[X] Bug

OPpuolitaival, 2018-09-13:
This is pure icetea issue. Please report in right place: https://github.com/ARMmbed/icetea 
Seems easy to fix

adbridge, 2018-09-14:
Needs an update to mbed-os for requirements.txt after icetea release

cmonr, 2018-09-17:
Fore reference: https://github.com/ARMmbed/mbed-os/pull/8140

Retesting needed once RC3 is generated.

MarceloSalazar, 2018-09-19:
Fixed in RC3, Icetea 1.0.2 :)

closed 2018-09-19 by MarceloSalazar