ARMmbed/mbed-os #8120: Icetea must be run from the mbed-os root

SeppoTakalo, 2018-09-13:
### Description

Apparently this method of running tests does not work anymore
```
mbed new test
cd test
mbed test -t <target> -m <toolchain>
```

In order to run Icetea tests, it seems to require that you are inside of Mbed OS folder.

```
$ mbed test --icetea --test-config tools/test_configs/HeapBlockDeviceAndEthernetInterface.json -v -m K66F -t GCC_ARM
[mbed] Working path "/Users/septak01/src/tmp/socket-test" (program)
[mbed] Exec "/Users/septak01/src/virtualenvs/mbed-build/bin/python2.7 -u /Users/septak01/src/tmp/socket-test/mbed-os/tools/run_icetea.py -m K66F -t GCC_ARM -v --application-list" in "/Users/septak01/src/tmp/socket-test"
[mbed] ERROR: "/Users/septak01/src/virtualenvs/mbed-build/bin/python2.7" returned error.
       Code: 1
       Path: "/Users/septak01/src/tmp/socket-test"
       Command: "/Users/septak01/src/virtualenvs/mbed-build/bin/python2.7 -u /Users/septak01/src/tmp/socket-test/mbed-os/tools/run_icetea.py -m K66F -t GCC_ARM -v --application-list"
       Tip: You could retry the last command with "-v" flag for verbose output
---

$ cd mbed-os
$ mbed test --icetea --test-config tools/test_configs/HeapBlockDeviceAndEthernetInterface.json -v -m K66F -t GCC_ARM
...
Build successes:
  * K66F::GCC_ARM::MBED-BUILD
  * K66F::GCC_ARM::TEST_APPS-DEVICE-SOCKET_APP

...
17:46:51.163 Suite creation complete.
17:46:51.193 Test cases prepared.
+-------+--------------------------------+----------+-------+---------+---------------------+--------------------------+---------+-------------------+
| Index |              Name              |  Status  |  Type | Subtype |        Group        |        Component         | Feature | Allowed platforms |
+-------+--------------------------------+----------+-------+---------+---------------------+--------------------------+---------+-------------------+
| 1     |          test_cmdline          | released | smoke |         |  testcases/example  |       ['cmdline']        |         |                   |
| 2     |      UDPSOCKET_BIND_PORT       | released | smoke |  socket | testcases/netsocket | ['mbed-os', 'netsocket'] |         |                   |
| 3     |      TCPSOCKET_BIND_PORT       | released | smoke |  socket | testcases/netsocket | ['mbed-os', 'netsocket'] |         |                   |
| 4     |        TCPSERVER_ACCEPT        | released | smoke |  socket | testcases/netsocket | ['mbed-os', 'netsocket'] |         |                   |
| 5     | TCPSOCKET_ECHOTEST_BURST_SHORT | released | smoke |  socket | testcases/netsocket | ['mbed-os', 'netsocket'] |         |                   |
+-------+--------------------------------+----------+-------+---------+---------------------+--------------------------+---------+-------------------+

```

So it starts to run the testcases, but only if I start from mbed-os folder.

Is this documented behaviour?
@OPpuolitaival 
Might be a root cause for #8064 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

OPpuolitaival, 2018-09-13:
@SeppoTakalo Do you know do we want to run icetea tests in subfolders? Is it good to make same solution than unittests that just quit with warning when running that in folder that does not contain TEST_APPS folder.

OPpuolitaival, 2018-09-13:
Actually the question is that do we want this to be like unittest or greentea?

Unittest: print warning and quit if no test folder in the current directory
Greentea: Walk trough the folder structure and run all tests that it finds

SeppoTakalo, 2018-09-14:
Hmm...
Actually, I'm unsure now of which is the most used and documented way.

Our PR CI seems to run `mbed test` from mbed-os folder.
So if this is the way it should be ran, then it has to be properly documented.

SeppoTakalo, 2018-09-14:
I can find both form from our documentation:
* `mbed test -t <toolchain> -m <target> -n "tests-mbed_hal-sleep*"`
* `mbed test -t <toolchain> -m <target> -n mbed-os-tests-mbed_hal*`

So apparently it is not unified.

In which case, I would perhaps launch a warning if TEST_APPS is not found.. that might be the cleanest solution.

OPpuolitaival, 2018-09-14:
@SeppoTakalo I will implement warning. I think that we can change it later because change is not breaking anything

SeppoTakalo, 2018-09-14:
This issue will be fixed by https://github.com/ARMmbed/mbed-cli/pull/758

cmonr, 2018-09-17:
Fixed, but will need to be released in 1.8.2 of Mbed CLI, and retested.

OPpuolitaival, 2018-10-02:
I think that this is resolved now

closed 2018-10-02 by adbridge