ARMmbed/mbed-os #8126: Getting hardfault when trying to run netsocket tests on wifi boards

aashishc1988, 2018-09-13:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
DISCO_L475VG_IOT01A

**Toolchain:**
ARM

**Tests we are failing:**
TCPSERVER_ACCEPT  (Icetea)
TCPSOCKET_ECHOTEST_BURST_SHORT

**Steps to reproduce**

mbed test -m DISCO_L475VG_IOT01A -t ARM --icetea -n TCPSERVER_ACCEPT

**Modifications made to enable the test to build and run"
Created mbed_app.json that looked like

{
    "config": {
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"iotlab\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"42F67YxLX4AawRdcj\""
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
			"target.network-default-interface-type": "WIFI",
            "nsapi.default-wifi-ssid": "\"iotlab\"",
            "nsapi.default-wifi-password": "\"PASSWORD_HIDDEN_OBVIOUSLY\""
        }
    },
	"macros": ["MBED_EXTENDED_TESTS"]
}

**log snippet**
```
> 
cleaning up Icetea log directory.
14:49:54.247 Using Icetea version 1.0.1
14:49:54.247 Loading suite from file
14:49:54.247 Importing testcases for filtering
14:49:54.309 Filtering testcases
14:49:54.309 Suite creation complete.
14:49:54.315 Test cases prepared.
14:49:54.315 Starting suite.
14:49:54.325 Starting repeat 1 of 1
14:49:54.325 Iteration 1 of 1
14:49:54.325
14:49:54.325 START TEST CASE EXECUTION: 'TCPSERVER_ACCEPT'
14:49:54.335
14:49:54.335 | TC     MainThread: Start Test case 'TCPSERVER_ACCEPT'
14:49:54.335 | TC     MainThread: ====setUpTestBench====
14:49:54.348 | TC     MainThread: ====setUpTestBench====
14:49:54.348 | TC     MainThread: ---------------------
14:49:54.348 | TC     MainThread: Initialize DUT's connections
14:49:54.348 | RSP     MainThread: Allocating 2 duts.
14:49:54.404 | LAL     MainThread: Allocated device 076402216E4665725102F32B
14:49:54.414 | LAL     MainThread: Allocated device 076402216E486572510CF32B
[... 117 lines cut ...]
14:52:42.183 | D2 --> DutThread: set --vt100 on
14:52:42.223 | D2 <-- DutThread: retcode: 0
14:52:42.243 | D2 --> DutThread: set --retcode false
14:52:42.305 | TC     MainThread: Test 'TCPSERVER_ACCEPT' FAIL, reason: COM40 CMD timeout: ifup
14:52:42.723 Test case TCPSERVER_ACCEPT failed, No retries left.

+------------------+---------+-------------------------+-------------+---------------------+----------+
| Testcase         | Verdict |       Fail Reason       | Skip Reason |      platforms      | duration |
+------------------+---------+-------------------------+-------------+---------------------+----------+
| TCPSERVER_ACCEPT |   fail  | COM40 CMD timeout: ifup |             | DISCO_L475VG_IOT01A |  167.97  |
+------------------+---------+-------------------------+-------------+---------------------+----------+
+---------------+----------------+
|    Summary    |                |
+---------------+----------------+
| Final Verdict |      FAIL      |
|     count     |       1        |
|    passrate   |     0.00 %     |
|      fail     |       1        |
|    Duration   | 0:02:47.970000 |
+---------------+----------------+
14:52:42.775 Cleanup done.
```
    [ ] Question
    [ ] Enhancement 
    [X ] Bug

aashishc1988, 2018-09-13:
@SeppoTakalo @OPpuolitaival @jarlamsa

jeromecoutant, 2018-09-14:
Hi
Just quick questions to start:
- did you well import https://github.com/ARMmbed/wifi-ism43362 ?
- In my own mbed_app.json file, I have also "nsapi.default-wifi-security": "WPA2",
- I don't know what is icetea, is it working without it ?

deepikabhavnani, 2018-09-14:
@jarlamsa - Can we update the readme stating default configuration is NONE and mbed_app.json cannot be used at same time with test configurations..

aashishc1988, 2018-09-14:
@jarlamsa I followed the instructions, that is,  added wifi information in ISM43362Interface.json, and in target_configs.json, I changed default configuration to ISM43362_WIFI, now i get the following error

```
:\testCheckMbedOS\mbed-os>mbed test -m DISCO_L475VG_IOT01A -t ARM --icetea
[mbed] Auto-installing missing Python modules...
[ERROR] Attempt to override undefined parameter 'ism43362.provide-default' in 'application[*]'
Traceback (most recent call last):
  File "C:\testCheckMbedOS\mbed-os\tools\run_icetea.py", line 303, in <module>
    raise Exception("Build data file does not exist: {}".format(options.build_data))
Exception: Build data file does not exist: .\BUILD\tests\DISCO_L475VG_IOT01A\ARM\build_data.json
[ERROR] Build data file does not exist: .\BUILD\tests\DISCO_L475VG_IOT01A\ARM\build_data.json
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\testCheckMbedOS\mbed-os"
       Command: "c:\python27\python.exe -u C:\testCheckMbedOS\mbed-os\tools\run_icetea.py -m DISCO_L475VG_IOT01A -t ARM --build-data .\BUILD\tests\DISCO_L475VG_IOT01A\ARM\build_data.json --test-suite .\BUILD\tests\DISCO_L475VG_IOT01A\ARM\test_suite.json"
       Tip: You could retry the last command with "-v" flag for verbose output
```

Is there some other step that is missing?

jarlamsa, 2018-09-17:
```
[ERROR] Attempt to override undefined parameter 'ism43362.provide-default' in 'application[*]'
```
Seems like you don't have the ISM43362 driver, please add the driver to the mbed-os folder by running:
```
mbed add wifi-ism43362
```

aashishc1988, 2018-09-17:
Thanks @jarlamsa . We should modify the documentation to include that, and also adding

```
"nsapi.default-wifi-security": "WIFI_SECURITY",
```
in ISM43362Interface.json gives a build error that some macro (I have to build again to remember the name) is not defined so I had to get rid of that. Now I can get it to build and run, and it fails. Here is the log. May be I should close this issue and file a new one.

```
cleaning up Icetea log directory.
08:58:38.628 Using Icetea version 1.0.1
08:58:38.628 Loading suite from file
08:58:38.634 Importing testcases for filtering
08:58:38.684 Filtering testcases
08:58:38.694 Suite creation complete.
08:58:38.698 Test cases prepared.
08:58:38.704 Starting suite.
08:58:38.704 Starting repeat 1 of 1
08:58:38.704 Iteration 1 of 1
08:58:38.704
08:58:38.714 START TEST CASE EXECUTION: 'UDPSOCKET_BIND_PORT'
08:58:38.714
08:58:38.714 | TC     MainThread: Start Test case 'UDPSOCKET_BIND_PORT'
08:58:38.714 | TC     MainThread: ====setUpTestBench====
08:58:38.724 | TC     MainThread: ====setUpTestBench====
08:58:38.724 | TC     MainThread: ---------------------
08:58:38.724 | TC     MainThread: Initialize DUT's connections
08:58:38.734 | RSP     MainThread: Allocating 1 duts.
08:58:38.865 | LAL     MainThread: Allocated device 076402216E4665725102F32B
08:58:38.865 | RSP     MainThread: Allocation successful
08:58:38.875 | D1     Thread-5: Flashing dev: 076402216E4665725102F32B
08:58:38.964 | FLS     Thread-5: sendBreak to device to reboot
08:58:39.224 | FLS     Thread-5: reset completed
        1 file(s) copied.
08:58:47.404 | FLS     Thread-5: sendBreak to device to reboot
08:58:47.654 | FLS     Thread-5: reset completed
08:58:48.062 | FLS     Thread-5: 076402216E4665725102F32B flash success
08:58:48.065 | TC     Thread-5: flash ready
08:58:48.065 | TC     Thread-5: Use board DISCO_L475VG_IOT01A as "D1" (id: 076402216E4665725102F32B)
08:58:48.134 | D1     MainThread: DutSerial COM40

+----------+--------------+
| COM port |   Location   |
+----------+--------------+
|  COM40   | X = 0, Y = 0 |
+----------+--------------+
08:58:48.141 | D1 <-> MainThread: Open Connection for 'D1' using 'COM40' baudrate: 115200
08:58:48.155 | D1 <-> MainThread: Use normal serial write mode
08:58:48.157 | TC     MainThread: Waiting for dut 1 to initialize.
08:58:48.158 | D1 --> DutThread: set --retcode true
/>set --retcode true  DutThread: />s
08:58:48.207 | D1 <-- DutThread: retcode: 0
08:58:48.227 | D1 --> DutThread: echo off
/>echo off 6 | D1 <-- DutThread: />
08:58:48.276 | D1 <-- DutThread: ECHO is off
08:58:48.296 | D1 <-- DutThread: retcode: 0
08:58:48.315 | D1 --> DutThread: set --vt100 off
08:58:48.335 | D1 <-- DutThread: retcode: 0
08:58:48.342 | D1 --> DutThread: set dut "COM40"
08:58:48.355 | D1 <-- DutThread: retcode: 0
08:58:48.361 | D1 --> DutThread: set testcase ""
08:58:48.378 | D1 <-- DutThread: retcode: 0
08:58:48.380 | TC     MainThread: ------TC SET UP---------
08:58:48.381 | D1 --> DutThread: ifup
08:58:50.976 | D1 <-- DutThread: retcode: -1
08:58:51.036 | TC     MainThread: command fails - set preliminaryVerdict as FAIL
08:58:51.036 | TC     MainThread: Test step fails!
08:58:51.036 | TC     MainThread: Test Case fails because of: dut1 cmd: 'ifup',retcode: -1
08:58:51.036 | TC     MainThread: Skip TC case
08:58:51.036 | TC     MainThread: ====TC TEAR DOWN====
08:58:51.036 | D1 --> DutThread: ifdown
08:58:51.046 | D1 <-- DutThread: retcode: 0
08:58:51.049 | TC     MainThread: ====tearDownTestBench====
08:58:51.052 | D1 --> DutThread: echo on
08:58:51.068 | D1 <-- DutThread: ECHO is on
08:58:51.078 | D1 <-- DutThread: retcode: 0
08:58:51.086 | D1 --> DutThread: set --vt100 on
08:58:51.118 | D1 <-- DutThread: retcode: 0
08:58:51.137 | D1 --> DutThread: set --retcode false
08:58:51.196 | TC     MainThread: Test 'UDPSOCKET_BIND_PORT' FAIL, reason: dut1 cmd: 'ifup',retcode: -1
08:58:51.936 Test case UDPSOCKET_BIND_PORT failed, No retries left.

+---------------------+---------+------------------------------+-------------+---------------------+----------+
| Testcase            | Verdict |         Fail Reason          | Skip Reason |      platforms      | duration |
+---------------------+---------+------------------------------+-------------+---------------------+----------+
| UDPSOCKET_BIND_PORT |   fail  | dut1 cmd: 'ifup',retcode: -1 |             | DISCO_L475VG_IOT01A |  12.481  |
+---------------------+---------+------------------------------+-------------+---------------------+----------+
+---------------+----------------+
|    Summary    |                |
+---------------+----------------+
| Final Verdict |      FAIL      |
|     count     |       1        |
|    passrate   |     0.00 %     |
|      fail     |       1        |
|    Duration   | 0:00:12.481000 |
+---------------+----------------+
08:58:51.956 Cleanup done.
```

I will look into the test to see what that command does, it just seems may be it not able to connect to wifi?

jarlamsa, 2018-09-18:
It means that it wasn't able to connect to the wifi network.
In this `"nsapi.default-wifi-security": "WIFI_SECURITY",` the `"WIFI_SECURITY"` should be replaced by the encryption protocol the access point is using, list of supported encryption protocols can be found from mbed-os/features/netsocket/nsapi_types.h, currently they include WEP, WPA, WPA2, WPA_WPA2, PAP and CHAP

Yes the documentation should be improved somewhat, but in a generic way.

aashishc1988, 2018-09-18:
@jarlamsa Thank you for clarifying that. :) The original issue listed here does not exist anymore so I would close this one.

closed 2018-09-18 by aashishc1988