ARMmbed/mbed-os #8143: mbed dm - SyntaxError as async keyword being

MarceloSalazar, 2018-09-17:
### Description

Evaluating 5.10 (RC2) and the Pelion device management feature.
I'm seeing Python errors as the keyword 'async' is being used:

```
$ mbed dm init -d "arm.com" --model-name "mbed" -v
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal" (program)
[mbed] Auto-installing missing Python modules...
[mbed] Exec "/usr/local/opt/python/bin/python3.7 -u /Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/device_management.py init -d arm.com --model-name mbed" in "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal"
Traceback (most recent call last):
  File "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/device_management.py", line 30, in <module>
    from mbed_cloud import AccountManagementAPI, CertificatesAPI
  File "/usr/local/lib/python3.7/site-packages/mbed_cloud/__init__.py", line 22, in <module>
    from mbed_cloud.account_management import AccountManagementAPI
  File "/usr/local/lib/python3.7/site-packages/mbed_cloud/account_management/__init__.py", line 1, in <module>
    from .account_management import AccountManagementAPI
  File "/usr/local/lib/python3.7/site-packages/mbed_cloud/account_management/account_management.py", line 29, in <module>
    from mbed_cloud._backends import iam
  File "/usr/local/lib/python3.7/site-packages/mbed_cloud/_backends/iam/__init__.py", line 53, in <module>
    from .apis.account_admin_api import AccountAdminApi
  File "/usr/local/lib/python3.7/site-packages/mbed_cloud/_backends/iam/apis/__init__.py", line 4, in <module>
    from .account_admin_api import AccountAdminApi
  File "/usr/local/lib/python3.7/site-packages/mbed_cloud/_backends/iam/apis/account_admin_api.py", line 137
    async=params.get('async'),
        ^
SyntaxError: invalid syntax
[mbed] ERROR: "/usr/local/opt/python/bin/python3.7" returned error.
       Code: 1
       Path: "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal"
       Command: "/usr/local/opt/python/bin/python3.7 -u /Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/device_management.py init -d arm.com --model-name mbed"
       Tip: You could retry the last command with "-v" flag for verbose output
```

Environment OSX El Capital
Mbed CLI 1.8.1
Python 3.7

Environement
[ ] Question  
[ ] Enhancement  
[X] Bug

teetak01, 2018-09-17:
Its possible that this is fixed in SDK version 2.0.1

MarceloSalazar, 2018-09-17:
@teetak01 you're correct.
We should update this line https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.10/requirements.txt#L19

cmonr, 2018-09-18:
Will be fixde with https://github.com/ARMmbed/mbed-os/pull/8160

closed 2018-09-20 by cmonr