ARMmbed/mbed-os #8146: Error at compile MCC app with K64F bootloader

MarceloSalazar, 2018-09-17:
### Description

Evaluating 5.10 (RC2) and the Pelion device management feature with mbed-cloud-client-example-sources-internal app (sorry internal only).

I'm seeing an error at the last step of the compile process (merge with bootloader)

```
mbed import mbed-cloud-client-example-sources-internal
cd mbed-cloud-client-example-sources-internal
mbed update 2.0.0-RC2
<CLOUD_SDK_API_KEY already set>
mbed dm init -d "arm.com" --model-name "mbed" -v -q --force
...
Merging Regions
  Filling region bootloader with /Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/features/FEATURE_BOOTLOADER/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/mbed-bootloader-k64f-block_device-sotp-v3_4_0.bin
  Padding region bootloader with 0x9a4 bytes
Traceback (most recent call last):
  File "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/make.py", line 293, in <module>
    ignore=options.ignore
  File "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/build_api.py", line 548, in build_project
    merge_region_list(region_list, res, notify)
  File "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/build_api.py", line 423, in merge_region_list
    _fill_header(region_list, region).tofile(header_filename, format='hex')
  File "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/build_api.py", line 380, in _fill_header
    header.puts(start, struct.pack(fmt, time()))
struct.error: required argument is not an integer
[mbed] ERROR: "/usr/local/opt/python/bin/python3.7" returned error.
       Code: 1
       Path: "/Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal"
       Command: "/usr/local/opt/python/bin/python3.7 -u /Users/ms/mbed/OOB/5.10/mbed-cloud-client-example-sources-internal/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -c -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

This makes me think there is a problem with the bootloader, the tools or the app.

[ ] Question  
[ ] Enhancement  
[X] Bug  

@theotherjimmy @teetak01 can you please have a look and comment?

teetak01, 2018-09-17:
I am unable to reproduce this. Might this be related to specific pip components versions or did you do any changes to mbed_app.json? I tested with python-sdk 2.0.0 and 2.0.1

MarceloSalazar, 2018-09-17:
I can't reproduce on Win7 / Python2.7.
Maybe a problem with Python 3?

MarceloSalazar, 2018-09-17:
I'm able to reproduce this on Windows 10 as well with Python 3.6.5

vijhar01, 2018-09-18:
I am able to reproduce this on Windows 10 with Python 3.6.3:
( Version - mbed update to 2.0.0-RC2 ran without any errors)
----------------------------------------------------------------------------
PS C:\oob\mbed-cloud-client-example-sources-internal> mbed compile --app-config configs/eth_v4.json
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling,
 testing and exporting
---
[mbed] Auto-installing missing Python modules...
[mbed] Detected "NUCLEO_F429ZI" connected to "D:" and using com port "COM4"
Building project mbed-cloud-client-example-sources-internal (NUCLEO_F429ZI, IAR)
......
Compile [ 99.8%]: update_ui_example.cpp
Compile [ 99.9%]: mcc_common_button_and_led.cpp
Compile [100.0%]: mcc_common_setup.cpp
Link: mbed-cloud-client-example-sources-internal_application
Elf2Bin: mbed-cloud-client-example-sources-internal_application
Merging Regions
  Filling region bootloader with C:\oob\mbed-cloud-client-example-sources-internal\mbed-os\features\FEATURE_BOOTLOADER\t
argets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F429xI\TARGET_NUCLEO_F429ZI\mbed-bootloader-nucleo_f429zi-block_device-sotp
-v3_4_0.bin
  Padding region bootloader with 0x2ec bytes
[ERROR] required argument is not an integer
[mbed] ERROR: "c:\program files (x86)\python36-32\python.exe" returned error.
       Code: 1
       Path: "C:\oob\mbed-cloud-client-example-sources-internal"
       Command: "c:\program files (x86)\python36-32\python.exe -u C:\oob\mbed-cloud-client-example-sources-internal\mbed
-os\tools\make.py -t iar -m NUCLEO_F429ZI --source . --build .\BUILD\NUCLEO_F429ZI\IAR --app-config configs/eth_v4.json"

       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---

vijhar01, 2018-09-19:
Above reported error encountered in RC3 build as well

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-458*

bridadan, 2019-02-21:
This was fixed last year in 5.10.2 (tried it locally to confirm): https://github.com/ARMmbed/mbed-os/pull/8330

closed 2019-02-21 by cmonr