ARMmbed/mbed-os #8150: U-blox cellular PPP driver is missing APN authentication

AriParkkila, 2018-09-17:
### Description

[U-blox cellular PPP driver](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular/framework/targets/UBLOX/PPP) is missing implementation for PDP context authentication similar to AT+CGAUTH command in 3GPP TS 27.007.

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

screamerbg, 2018-09-26:
CC @RobMeades

RobMeades, 2018-09-26:
Looking at 27.007 section 10.1.31 `AT+CGAUTH` is an optional AT command and, since it doesn't appear in the [u-blox AT command manual](https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf) I'm afraid that no u-blox *module* implements it; this is nothing to do with the driver, the modules do not support `AT+CGAUTH`.  We do have a proprietary command, `AT+UAUTHREQ` (section 18.39 of the manual) which does the same thing as `AT+CGAUTH` but we don't expect anyone to use it.  The expected mode of operation is that the APN is specified in the usual way with the `AT+CGDCONT` command and any associated username and password are carried in the normal way through the `PPP` setup process (see the note in section 18.39.4 of the manual).  So the username and password would go into your PPP stack instead of being sent separately via an AT command.  Is there a reason why this would not work?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCELL-1415*

closed 2018-11-30 by ciarmcom