ARMmbed/mbed-os #8151: Cannot find DataFlashBlockDevice tests in Mbed OS 5.10

korjaa, 2018-09-17:
### Description
dataflash-driver seems to be moved inside Mbed OS in version 5.10 but I cannot find the tests anywhere for it.

Namely this file
https://github.com/ARMmbed/dataflash-driver/blob/master/TESTS/block_device/dataflash/main.cpp

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

davidsaada, 2018-09-17:
@yossi2le Please handle.

yossi2le, 2018-09-20:
@korjaa 
Sorry for that but unfortunately, Some of the tests in the external drivers repositories are protected by GNU license which cannot be used in mbed-os. Therefore, we have removed the tests which had the license issue and we will take care for it in the future by creating new tests for blockdevice.

0xc0170, 2018-09-20:
@korjaa I believe this is answered, and @yossi2le - do we have a tracking issue for this, to reference it so this can be closed?

yossi2le, 2018-09-20:
@0xc0170 We have a task at the JIRA 
IOTSTOR-658

closed 2018-09-20 by 0xc0170