ARMmbed/mbed-os #8152: GCC 7.1.0 support - mbed-cli warning

drahnr, 2018-09-17:
### Description
I am currently compiling mbed os with GCC 7.1.0 in CI and I get a warning regarding unsupported compilers.
Is thise warning due to a known issue or due to a lack of testing?

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-09-17:
The latter, lack of testing (not yet supported).

drahnr, 2018-09-17:
What is required for verification of a new GCC compiler version?

drahnr, 2018-09-18:
@mattbrown015 just making sure I am not running into issues that are known, just because my GCC is 7.x.y compared to 6.x.y

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1642*

cmonr, 2019-01-31:
@drahnr The message was meant to make it clear what we explicitly support and test against.

A bit after we added the check, we downgraded the message from an error to a warning: https://github.com/ARMmbed/mbed-os/pull/7876

Since I think that answers your question, I'll be closing this. If it doesn't, feel free to reopen.

closed 2019-01-31 by cmonr