ARMmbed/mbed-os #8158: Not all components are conditionally compiled

sg-, 2018-09-17:
### Description
There are components that are not following the COMPONENT_ pattern to be conditionally compiled, added and removed.

Currently this is the case for:
- ```wifi/esp8266-driver```
- ```802.15.4_RF/<everything underneath it>```

This is correctly implemented for ```storage/blockdevice```

Also linking similar issues that relate to cellular drivers not in components but could make good use of the capability. #7457 

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-09-18:
@ARMmbed/mbed-os-ipcore Please review

SeppoTakalo, 2018-09-18:
What problem does the conditional compilation fix?

All drivers we submitted, are able to compile on all platforms. Conditional compilation just add one layer of complexity to user configuration.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/ONME-3893*

SeppoTakalo, 2019-02-06:
New compilation profiles are being designed, so we will recheck this topic once the design is ready.
See internal ticket https://jira.arm.com/browse/ISGDEVPREQ-1003

closed 2019-02-06 by ciarmcom