ARMmbed/mbed-os #8170: Networking overrides - stop compiling if no pins defined

JanneKiiskila, 2018-09-18:
### Description

As the network interfaces can be overridden and all boards do not have pins, there are not always any defaults - pins are directed to "NC". This means, that if you don't exactly know WHAT you must configure, you can easily build a config that (assuming from previous experience with `easy-connect` with working default pins) that is actually mapping TX/RX to "NC". This means you produce a binary that completely fails to work and it's actually quite difficult to debug why it failed.

The compilation should fail, if you have not mapped your pins to something else than "NC".

For ESP8266 for. ex. we have to provide in `mbed_app.json`:

```
            "esp8266.rx"                : "D0",
            "esp8266.tx"                : "D1",
            "esp8266.provide-default"   : true,
```
If you leave the rx/tx out, you will spend time debugging the issue... Better fail quickly, already in compilation phase.

### Issue request type
    [ ] Question
    [x] Enhancement 
    [ ] Bug 

Kudos to @teetak01 for finding this issue.

JanneKiiskila, 2018-09-18:
Attn: @SeppoTakalo  @kjbracey-arm

SeppoTakalo, 2018-09-18:
Yes.
I agree, we should do it for drivers that expect pins to be defined.

kjbracey-arm, 2018-09-18:
Not sure what the exact failure mode currently is - in this particular case I would imagine the attempt to have "provide default" true and no pins should probably be a compilation failure, yes.

If provide default was false, I would expect the "NetworkInterface::get_default_instance" to just return NULL, which would be a clear enough "no network interface available".

Looking more closely - yes, the default constructor of `ESP8266Interface` is not useful if the pins aren't defined. I would suggest the code should omit the default constructor if the pins aren't set and that would make the "get_default" fail to compile, causing an error if the "provide-default" was set.

The area was discussed on https://github.com/ARMmbed/esp8266-driver/pull/75 - my main concern then was avoiding build breakage when the driver is not in use, and also making it still work if you're manually specifying pins in C++.

kjbracey-arm, 2018-09-18:
Although on this specific case - the driver is supposed to default to `D0` and `D1`, if you're Arduino form-factor, so you shouldn't actually need to specify the pins in a lot of cases.

kjbracey-arm, 2018-09-18:
And a final note - literal use of `"NC"` in the JSON would be a headache, as we can't easily test for it at compile time, because it's an enum, not a define. I believe the driver is now set up so that none is represented as `null` in the JSON, which can be checked with an `#ifdef`

SeppoTakalo, 2018-09-18:
Can you still test it like this?
```
#define NC 12345
#if MBED_CONF_ESP8266_TX == NC
#error "[error] Pin is not defined"
#endif
#undef NC
```

kjbracey-arm, 2018-09-18:
Yes, but not sure that passes the "easily" test.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/ONME-3898*

closed 2018-10-23 by ciarmcom