ARMmbed/mbed-os #8172: Traceback error stderr.decode() when running Icetea with Python 3

MarceloSalazar, 2018-09-18:
### Description

Evaluating Mbed OS 5.10 RC3 and seeing traceback error:

```
mbed test -m K64F -t GCC_ARM --icetea -vv --compile-list
...
[mbed-50001] Exec "/usr/local/opt/python/bin/python3.7 -u /Users/ms/mbed/OOB/5.10/rc3/mbed-os/tools/run_icetea.py -m K64F -t GCC_ARM -v --compile-list" in "/Users/ms/mbed/OOB/5.10/rc3"
Traceback (most recent call last):
  File "/Users/ms/mbed/OOB/5.10/rc3/mbed-os/tools/run_icetea.py", line 278, in <module>
    icetea_json_output = icetea_tests(options.target, options.tcdir, options.verbose)
  File "/Users/ms/mbed/OOB/5.10/rc3/mbed-os/tools/run_icetea.py", line 177, in icetea_tests
    stderr.decode()))
Exception: Error when running icetea. 
cwd:/Users/ms/mbed/OOB/5.10/rc3 
Command:'icetea --tcdir TEST_APPS --list --json --platform_filter K64F -v' 
output:19:04:21.105 Registering execution wide plugins:
19:04:21.105 Registering plugin LocalAllocator
19:04:21.105 Execution wide plugins loaded and registered.
19:04:21.189 Using Icetea version 1.0.2
19:04:21.189 Generating suite from command line.
19:04:21.190 Importing testcases for filtering
19:04:21.190 Test case directory does not exist!
19:04:21.190 Something went wrong in suite creation! Suite creation failed.

[ERROR] Error when running icetea. 
cwd:/Users/ms/mbed/OOB/5.10/rc3 
Command:'icetea --tcdir TEST_APPS --list --json --platform_filter K64F -v' 
output:19:04:21.105 Registering execution wide plugins:
19:04:21.105 Registering plugin LocalAllocator
19:04:21.105 Execution wide plugins loaded and registered.
19:04:21.189 Using Icetea version 1.0.2
19:04:21.189 Generating suite from command line.
19:04:21.190 Importing testcases for filtering
19:04:21.190 Test case directory does not exist!
19:04:21.190 Something went wrong in suite creation! Suite creation failed.

[mbed-50001] ERROR: "/usr/local/opt/python/bin/python3.7" returned error.
       Code: 1
       Path: "/Users/ms/mbed/OOB/5.10/rc3"
       Command: "/usr/local/opt/python/bin/python3.7 -u /Users/ms/mbed/OOB/5.10/rc3/mbed-os/tools/run_icetea.py -m K64F -t GCC_ARM -v --compile-list"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed-50001] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
```

Environment OSX El Capitan
- Mbed CLI 1.8.1
- Python 3.7
- Icetea 1.0.2

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

jupe, 2018-09-19:
looks that folder where `mbed test` does not contain `TEST_APPS` -folder, true? 
@OPpuolitaival does mbed-cli check if folder exists before spawning icetea?

jupe, 2018-09-19:
true, check is implemented here: 
https://github.com/ARMmbed/mbed-cli/pull/758. Is this something to do with python 3.7 🤔 Could you try to run icetea directly with same parameters to see if it works as standalone?

cmonr, 2018-09-19:
Hmm, stderr.decide appears to be a Python 3 string issue. I can take a look at it, but it will have to wait until tomorrow, as I'm wrapping up other things tonight.

OPpuolitaival, 2018-09-19:
@cmonr it is not actually decode problem.. error just coming from line where the exception is formatted and there decode is used. The decode itself works fine. I am working with this

OPpuolitaival, 2018-09-19:
@MarceloSalazar most probably you don't have TEST_APPS folder in a place where you run that command. There is better error handling for that in new mbed-cli version. I made same in mbed-os side https://github.com/ARMmbed/mbed-os/pull/8181 (and fixed one py3 issue)

MarceloSalazar, 2018-09-19:
I've realized this is a duplicate of 8064.
I'm happy to close this one and continue the conversation there.

closed 2018-09-19 by 0xc0170