ARMmbed/mbed-os #8173: NRF52832 (UBLOX NINA-B1) using BLE allowLegacyPairing not working as expected

edablan, 2018-09-18:
### Description

mbed 5.9.7 (949cb49ab)
UBLOX NINA-B1(NRF52832)
IAR 7.80.4
mbed-cli 1.7.2

I am setting up a BLE project and upon initialization of the securityManager later changing to use LegacyPairing. 

```C++
    ble.securityManager().init(true, false, SecurityManager::IO_CAPS_NONE, NULL, false, NULL);
    
    ble.securityManager().allowLegacyPairing(true);
    ble.securityManager().setKeypressNotification(false);
```

I do not want to use LE secure connection pairing, though using my BLE v4.2 sniffer I can see that secure connection pairing is being responded by my device even after using the alllowLegayPairing() function call. 

[image]

For curiosity I used only the Nordic SDK 14.2 (same as mbed) and ran some examples provided by Nordic that do not sure LE secure connection. Sure enough from the BLE v4.2 analysis tool I can see that LE secure connection is turned off. The nordic examples work at turning off LE secure connection pairing. 

All of my setSecurityRequirements for all characterisitcs used in my services are either NONE are UNAUTHENTICATED (which do not use LE secure connection).

I went further and attempted to force the lower level functions to turn off secure connections changing these lines of code found in GenericSecurityManager.cpp in the GenericSecurityManager::init() function

```C++
    bool secure_connections;
    _pal.get_secure_connections_support(secure_connections);

    _default_authentication.set_bondable(bondable);
    _default_authentication.set_mitm(mitm);
    _default_authentication.set_secure_connections(false); //<--- made this change to LE SC OFF!!
    _default_authentication.set_keypress_notification(true);
```
I used our BLE v4.2 analyzer and secure connections magically turns off. It works as expected. Not sure why this works all of a sudden.

[image]

Is there something I am missing in trying to turn off LE secure connections from my device side using the BLE API method? Or is this indicative of a bigger problem? Or is allowLegacyPairing(false) not the same as turning LE secure connections off? Is there a better way to turn off LE secure connections? Any help would be greatly appreciated.

### Issue request type

[X ] Question  
[ ] Enhancement  
[ ] Bug

screamerbg, 2018-09-25:
@pan-

pan-, 2018-09-25:
I believe it is not desirable to disable LESC pairing at the API level if supported by the hardware. 

However I do agree that in the current form it is not ideal for sniffing. One solution would be to have a configuration parameter that forces the use of the debug key (see BLUETOOTH SPECIFICATION Version 5.0 | Vol 3, Part H - 2.3.5.6.1). That solution as the advantage of **not** altering the link security level unlike using _Just Works_ pairing.

> Or is allowLegacyPairing(true) not the same as turning LE secure connections off?

Yes it is not the same as turning LESC off. It allows/disallows a LESC enabled controller to use or not use  the legacy pairing methods. 

------

Note: instead of changing the generic layer, you could modify your port to not exposes support of LESC: 

https://github.com/ARMmbed/mbed-os/blob/e43f0201a738ba64b41f07aa6a66b0d1d52da097/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NRF52/source/nRF5xPalSecurityManager.cpp#L366-L371

edablan, 2018-09-25:
@pan- 
Thanks pan I'll try that bit of code instead of inserting my line. I'll go ahead and close the ticket.

pan-, 2018-09-26:
@edablan I tried it on my end and it works as expected. I wonder if you have a source/compilation issue given you've two issues I cannot reproduce.

edablan, 2018-09-27:
@pan- 

I can confirm using a clean version of mbed-os-5.10.0 (610e35ddc) and changing the lines as suggested  works. Here was the suggested change:

```C++
ble_error_t nRF5xSecurityManager::get_secure_connections_support(
    bool &enabled
) {
    enabled = true;
    return BLE_ERROR_NONE;
}
```

closed 2018-09-27 by edablan