ARMmbed/mbed-os #8174: Issue with ARM_MPU_RASR macro for MPU setup

mmahadevan108, 2018-09-18:
### Description
The current implementation of the ARM_MPU_RASR macro inside cmsis/TARGET_CORTEX_M/mpu_armv7.h does not use the SubRegionDisable, Size fields. 

I reverted the ARM_MPU_RASR macro implementation to a previous commit 

#define ARM_MPU_RASR(DisableExec, AccessPermission, TypeExtField, IsShareable, IsCacheable, IsBufferable, SubRegionDisable, Size) \
-  ARM_MPU_RASR_EX(DisableExec, AccessPermission, ARM_MPU_ACCESS_(TypeExtField, IsShareable, IsCacheable, IsBufferable), SubRegionDisable, Size)
+      ((((DisableExec     ) << MPU_RASR_XN_Pos)     & MPU_RASR_XN_Msk)     | \
+       (((AccessPermission) << MPU_RASR_AP_Pos)     & MPU_RASR_AP_Msk)     | \
+       (((TypeExtField    ) << MPU_RASR_TEX_Pos)    & MPU_RASR_TEX_Msk)    | \
+       (((IsShareable     ) << MPU_RASR_S_Pos)      & MPU_RASR_S_Msk)      | \
+       (((IsCacheable     ) << MPU_RASR_C_Pos)      & MPU_RASR_C_Msk)      | \
+       (((IsBufferable    ) << MPU_RASR_B_Pos)      & MPU_RASR_B_Msk)      | \
+       (((SubRegionDisable) << MPU_RASR_SRD_Pos)    & MPU_RASR_SRD_Msk)    | \
+       (((Size            ) << MPU_RASR_SIZE_Pos)   & MPU_RASR_SIZE_Msk)   | \
+       (MPU_RASR_ENABLE_Msk))

The above change fixed the issue I was running into.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-09-19:
Fix is referenced above, please review

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-478*

evedon, 2020-05-15:
Closing since it seems that a fix was provided.

closed 2020-05-15 by evedon